prusa3d/PrusaSlicer #4874: Supports generated loose in space

rsurgiewicz, 2020-10-14:
### Version
PrusaSlicer-2.2.0+win64
PrusaSlicer-2.3.0-alpha0+win64

### Operating system type + version
Windows 10 64bit
Linux Arch (rolling)

### 3D printer brand / version + firmware version (if known)
Prusa MK3s / 3.9.1-3518

### Behavior
Supports for the part are generated hanging loose in space causing issues when printing.

* _Steps needed to reproduce the problem_
  * Load attached STL (`zaslepka.stl`) and orient it vertically
  * Select any profile 0.15 / 0.20 / 0.30 for Prusa mk3s (PREDEFINED, non-custom)
  * `Supports -> Everywhere`

  * .... or use included 3MF

* _Expected Results_:
    * Supports not generated loose in space.
* _Actual Results_
 [image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[3MF_and_STL.zip](https://github.com/prusa3d/PrusaSlicer/files/5376321/3MF_and_STL.zip)

Aquaerics, 2020-11-12:
I can confrim the Bug in PrusaSlicer-2.2.0+win64. Should be fixed soon. It is not possible to block the support.

I get the contact layer for slicing with "support only from the ground plate" for positions where it would be if i had support on all areas.

bubnikv, 2021-02-22:
PrusaSlicer 2.3.0 floating supports
[image]

Current master has these areas removed:
[image]

bubnikv, 2021-02-22:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/3985d7bb81ddd542612ec2e8704622be4e90f074

closed 2021-02-22 by bubnikv