prusa3d/PrusaSlicer #4880: prusa-gcodeviewer failes on Linux when used in AppImage and/or from File menu

Area5142, 2020-10-15:
### Version
PrusaSlicer-version_2.3.0-alpha0-1150-g58d57f9b7 and earlier versions

### Operating system type + version
Linux Mint 19.3 x64

### 3D printer brand / version + firmware version (if known)
NA.

### Behavior
Starting PrusaSlicer (PS) in gcode viewer mode fails when done from _File->G-code preview..._ or when binary is contained in AppImage.

**What is working:**
Sym-linking the compiled binary _prusa-slicer_ to _prusa-gcodeviewer_ and executing on build server. This starts PS in _gcode-view mode_ and gcode files can be loaded and viewed.

**What is not working:**
1) Using the menu _File->G-code preview..._ and opening gcode file. A splash screen is shown and dissapears again, PS is still running as nothing has happend. Loglevel=3 shows:

```
[2020-10-15 14:48:10.714408] [0x00007f4ab9fca5c0] [info]    Trying to spawn a new slicer "/home/.../src/PrusaSlicer/Build/src/prusa-gcodeviewer"

(prusa-gcodeviewer:118652): GLib-GObject-WARNING **: 14:48:10.832: invalid (NULL) pointer instance

(prusa-gcodeviewer:118652): GLib-GObject-CRITICAL **: 14:48:10.832: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
...
(prusa-gcodeviewer:118652): GLib-GObject-CRITICAL **: 14:48:10.846: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```
2) Starting PS with a symlink named _prusa-gcodeviewer_ to AppImage with binary. PS starts in normal PS mode and ignores the symlink name.

3) Trying to load/view gcode from menu _File->G-code preview..._ when binary is contained in an AppImage has no effect, but the the loglevel=3 output tells that PS tries to spawn a new process:

`[2020-10-15 15:04:54.077119] [0x00007fc868da75c0] [info]    Trying to spawn a new slicer "/home/.../bin/AppImages/PrusaSlicer-version_2.3.0-alpha0-1150-g58d57f9b7.AppImage"
`

There is no official AppImage build instruction to follow, but my builds has the same structure as earlier PS AppImages, just with updated libraries and works great as Slicer.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
NA.

bubnikv, 2020-10-15:
I would say please wait for the first public alpha release, where the structure of the AppImage will reveal including the start scripts and symlinks. The alpha is to be released any day now, keep your fingers crossed.

Area5142, 2020-10-15:
I will cross my finger and looks forward to see the changed structure of AppImage.

Even without the correct AppImage structure there is still a bug when the standalone prusa-slicer binary tries to execute the correctly named symlink prusa-gcodeviewer from menu File->G-code preview...
This call fails and is unrelated to any AppImage containment or structure.

I think this is still an issue and too soon to close.

bubnikv, 2020-10-16:
I tested with the yesterday's evening master, running on WSL2 (Windows
subsystem for Linux), and it works fine. This is the listing of my
build/src directory showing the symlinks.

❯ pwd
/home/bubnikv/src/PrusaSlicer/build/src
❯ ls -l
total 101320
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct  7 17:37 CMakeFiles
-rw-r--r-- 1 bubnikv bubnikv       631 Oct  7 17:37 CTestTestfile.cmake
-rw-r--r-- 1 bubnikv bubnikv      3715 Oct  7 17:37 Info.plist
-rw-r--r-- 1 bubnikv bubnikv       776 Oct  7 17:37
PrusaSlicer-gcodeviewer.rc
-rw-r--r-- 1 bubnikv bubnikv      1988 Oct  7 17:37 PrusaSlicer.manifest
-rw-r--r-- 1 bubnikv bubnikv       717 Oct  7 17:37 PrusaSlicer.rc
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 Shiny
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 admesh
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 avrdude
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 boost
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 build-utils
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 clipper
-rw-r--r-- 1 bubnikv bubnikv      3680 Oct  7 17:37 cmake_install.cmake
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 glu-libtess
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 hidapi
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 imgui
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 libigl
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 libnest2d
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:29 libslic3r
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 miniz
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct  7 17:37 platform
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 poly2tri
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 polypartition
lrwxrwxrwx 1 bubnikv bubnikv        12 Oct 15 10:30 prusa-gcodeviewer ->
prusa-slicer
-rwxr-xr-x 1 bubnikv bubnikv 103645184 Oct 15 10:30 prusa-slicer
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 qhull
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:27 semver
drwxr-xr-x 3 bubnikv bubnikv      4096 Oct 15 10:30 slic3r

Area5142, 2020-10-16:
To reproduce the error on Linux Mint 19.3:
1) Execute binary prusa-slicer after build.
2) Select File->G-code preview... and select a .gcode file.
3) The PS splash screen is shown and then nothing happens.

It is like the spawn process succeeded and the symlink is executed, but terminates before showing the gcode. I have tried several different gcodes and the result is the same.

The symlink prusa-gcodeviewer pointing to prusa-slicer works and starts a gcode viewer. I can load and view gcodes from the viewers File->Open G-code... menu and if given on the command line too.

bubnikv, 2020-10-16:
The differences I can think of:

1) Current path
2) LD_LIBRARY_PATH, environment and such
3) incorrect interpretation of symlinks when resolving the Resources
directory.

Frankly I am not sure what causes your issue.

Area5142, 2020-10-16:
The difference I see is that the prusa-gcodeviewer symlink is executed from inside prusa-slicer binary (via menu) instead of from the commend line.
The prusa-gcodeviewer starts and shows the splash screen, but then for some reason terminates itself again.
The environment setup should be the same as it will be inherited from the spawning process (prusa-slicer).

My guess would be that the spawn process in the startup code finds a lock, locked ressource, invalid parameters or detects the parent process and decides to terminate. I get no usable output from setting --loglevel=4.
A more detailed log of how/when the symlink is executed would share some light on, what is happening.

As mentioned before, executing the prusa-gcodeviewer link directly works as intended.

bubnikv, 2020-10-16:
Do you have "single instance" enabled?

This is a new settings to enable just a single instance of PrusaSlicer
running.

Area5142, 2020-10-16:
I found this new setting following the change-log.

No single instance is not enabled. I can start multiple instances of prusa-slicer and on top of them start multiple instances of prusa-gcodeviewer.

Area5142, 2020-10-16:
Maybe I am closer to find the error.
When I replace the prusa-gcodeviewer with a bash script, it is possible to write out the environment and how the symlink was called.

Starting prusa-gcodeviewer in different ways on the command line reviled some interesting results.

**Success:**
1) /home/joh/src/PrusaSlicer/Build/src/prusa-gcodeviewer ./gcode/Test.gcode
(Full path to binary and relativ path to gcode file)

**Fails:**
2) /home/joh/src/PrusaSlicer/Build/src/prusa-gcodeviewer /home/joh/src/PrusaSlicer/Build/src/gcode/Test.gcode
(Full path to binary and gcode file)

3) ./prusa-gcodeviewer /home/joh/src/PrusaSlicer/Build/src/gcode/Test.gcode
(Relative path to binary and full path gcode file)

4) ./prusa-gcodeviewer ./gcode/Test.gcode
(Relative path to binary and gcode file)

The prusa-gcodeviewer only starts with succes when given full path to binary and relative path to gcode file.

When prusa-gcodeviewer is started from PS it is given full path to both binary and gcode and fails like case 2.

This is the failed code written to the console:

```
(prusa-gcodeviewer:3578): GLib-GObject-CRITICAL **: 17:16:46.552: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Segmentation fault (core dumped)
```

The error may be located in start code of the gcode-viewer where the path to gcode is resolved. I could be a NULL pointer or other hard error that could give a segmentation fault - just my guess...

bubnikv, 2020-10-16:
/home/bubnikv/src/PrusaSlicer/build/src/prusa-gcodeviewer
/mnt/d/3dprint/20mmbox\ 20mmbox.gcode

both paths are full, it works nicely.

Are there any symlinks in the paths to the gcode viewer or to the g-code
file?

Would you please run the test under gdb, so you could list out a call stack
when it crashes?

Area5142, 2020-10-16:
My test machine is a clean Linux Mint 19.3 x64 image added with only enough to compile the source. Compile and test directory is is located on main disk with no spaces/symlinks in any path or file names.

I will try gdb to get a call stack.

Area5142, 2020-10-16:
Here the gdb session with fault, frame info and bt (back trace of stack):

[GDB_Stack_Dump.zip](https://github.com/prusa3d/PrusaSlicer/files/5392995/GDB_Stack_Dump.zip)

bubnikv, 2020-10-16:
For reference, I am copying the backtrace here.

It is really weird, it looks like it is crashing when sending to GPU. I
will ask Enrico to look into it, he may have an idea, however as of now the
chances are likely small.

Thread 1 "prusa-gcodeview" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) info frame
Stack level 0, frame at 0x7fffffff82f0:
 rip = 0x0; saved rip = 0x55555602e495
 called by frame at 0x7fffffff8320
 Arglist at 0x7fffffff82e0, args:
 Locals at 0x7fffffff82e0, Previous frame's sp is 0x7fffffff82f0
 Saved registers:
  rip at 0x7fffffff82e8
(gdb) bt
#0  0x0000000000000000 in  ()
#1  0x000055555602e495 in
Slic3r::GUI::GLModel::send_to_gpu(std::vector<float, std::allocator<float>
> const&, std::vector<unsigned int, std::allocator<unsigned int> > const&)
()
#2  0x000055555602e97e in
Slic3r::GUI::GLModel::init_from(Slic3r::GUI::GLModelInitializationData
const&) ()
#3  0x000055555603f864 in
Slic3r::GUI::GCodeViewer::SequentialView::Marker::init() ()
#4  0x00005555560443de in Slic3r::GUI::GCodeViewer::init() ()
#5  0x0000555556050eb5 in
Slic3r::GUI::GCodeViewer::load(Slic3r::GCodeProcessor::Result const&,
Slic3r::Print const&, bool) ()
#6  0x0000555555fcba0c in
Slic3r::GUI::GLCanvas3D::load_gcode_preview(Slic3r::GCodeProcessor::Result
const&) ()
#7  0x0000555556068820 in Slic3r::GUI::Preview::load_print_as_fff(bool) ()
#8  0x00005555560691fc in Slic3r::GUI::Preview::load_print(bool) ()
#9  0x0000555555d75235 in Slic3r::GUI::Plater::load_gcode(wxString const&)
()
#10 0x0000555555d335f6 in
Slic3r::GUI::GUI_App::AfterInitLoads::on_loads(Slic3r::GUI::GUI_App*) ()
#11 0x0000555555d3d5a7 in
wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>,
Slic3r::GUI::GUI_App::on_init_inner()::{lambda(wxIdleEvent&)#2}>::operator()(wxEvtHandler*,
wxEvent&) ()
#12 0x0000555556c0b0c1 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*,
wxEventFunctor&, wxEvent&) const (this=
    0x555557766920, handler=0x555557766920, functor=warning: RTTI symbol
not found for class 'wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>,
Slic3r::GUI::GUI_App::on_init_inner()::{lambda(wxIdleEvent&)#2}>'
..., event=...)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:671
#13 0x0000555556cab83a in
wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&,
wxEvtHandler*, wxEvent&) (entry=..., handler=0x555557766920, event=...)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
#14 0x0000555556cac503 in wxEvtHandler::SearchDynamicEventTable(wxEvent&)
(this=0x555557766920, event=...)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
#15 0x0000555556cabc80 in wxEvtHandler::TryHereOnly(wxEvent&)
(this=0x555557766920, event=...)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
#16 0x0000555556cad0a1 in wxEvtHandler::TryBeforeAndHere(wxEvent&)
(this=0x555557766920, event=...)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidge---Type
<return> to continue, or q <return> to quit---
[... 32 lines cut ...]
#28 0x0000555556c33c2d in wxEventLoopBase::Run() (this=0x555557f13ed0)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/evtloopcmn.cpp:90
#29 0x0000555556c0a684 in wxAppConsoleBase::MainLoop() (this=0x555557766920)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:380
#30 0x0000555556c0a41b in wxAppConsoleBase::OnRun() (this=0x555557766920)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:301
#31 0x00005555569f3893 in wxAppBase::OnRun() (this=0x555557766920)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidge---Type
<return> to continue, or q <return> to quit---
ts/src/common/appcmn.cpp:335
#32 0x0000555556c5ca53 in wxEntry(int&, wchar_t**) (argc=@0x555557640cf0:
2, argv=0x5555577667b0)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:507
#33 0x0000555556c5cb29 in wxEntry(int&, char**) (argc=@0x7fffffff9efc: 2,
argv=0x7fffffffdf88)
    at
/home/joh/src/PrusaSlicer/deps/Build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:519
#34 0x000055555585ce8d in Slic3r::CLI::run(int, char**) ()
#35 0x000055555583d18d in main ()
(gdb) Quit

bubnikv, 2020-10-17:
@lukasmatena notified me that he sees the G-code viewer sometimes opening below the terminal window or possibly PrusaSlicer window. I know there are issues with the order of OpenGL initialization on Linux, where we get OpenGL context initalized first at the first paint event. If the window is in the background, then we are possibly getting the paint event too late and accessing OpenGL crashes.

Area5142, 2020-10-17:
The gcode viewer only fails when started with a gcode file as parameter to the command. The splash screen is shown before it crashes.

Starting gcode viewer without gcode file as parameter works both as symlink and "prusa-slicer --gcodeviewer".

enricoturri1966, 2020-10-17:
@Area5142 
Could you please test if ce3fc31b488c0177fd5246ce4f2b4200955a51e2 helps in solving this issue ?
PrusaSlicer-version_2.3.0-alpha0+1190-gce3fc31b4

Area5142, 2020-10-17:
Quick test of the new alpha1.

Open gcode file from menu _File->G-code preview..._ in PS starts symlink to prusa-gcodeviewer and shows splash screen, shows small main windows and starts to load gcode. Then I get the following error popup:

[image]

If I start the symlink directly from command line with the same gcode file as parameter everything works.

bubnikv, 2020-12-05:
I suppose this has been fixed and could be closed?

Area5142, 2020-12-05:
Yes. Just tested with 2.3.0 beta 1 and it now works on Linux Mint 19.3 x64.

bubnikv, 2020-12-05:
Cool, thanks.

closed 2020-12-05 by bubnikv