prusa3d/PrusaSlicer #4881: Lost detail on high parts count slices.

nerobro, 2020-10-15:
### Version
2.2.0 (but I've seen in older versions too)

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3, Monoprice Mini Delta (however this problem happens before printing)

### Behavior
When attempting to fill the bed with parts, at some point, there are layer generation problems.  In a previous version I had some 100 small parts layed out on the bed, and the tips of them showed gaps.  The gaps appeared in the printed parts.  This time, I'm printing minis, and when I have more than about 40 items on the bed, upper sections of the parts have missing or errored layers.

Importantly, the parts layout looks good, and the render shows all the detail until the part is sliced.  

To replicate this problem, you need to put some large number of parts that use lots of segments on the build plate.  

The expected results are slices that look like the 3d rendered part.

The results I get are missing upper part details.  Gaps in curves, missing small details.  Gaps in general in the model.

Here's an image showing gaps, and missing details.  This has 63 parts on the build plate
[image]

Here's what the render looks like, before slicing.
[image]

This is what the sliced parts should look like: 
[image]

And they look good rendered as well:
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs[githubmissileRacks.zip]!(https://github.com/prusa3d/PrusaSlicer/files/5385436/githubmissileRacks.zip)

nerobro, 2020-10-15:
Oh yes, a symptom:  You will know when you've reached a sufficient parts count, when you go to slice, and the slicing reaches the calculating time part of the slicing process and.. stalls.  It can take hours for a large bed of parts to slice, and if it does, it definitely will have errors on the top layers.

rtyr, 2020-10-15:
I can see 3 things here.

**1) Non-manifold models, can be easily solved by repairing them.**
[image]
[image]

**2) Very long slicing time, caused by "avoid crossing perimeters" option.**
Known issue with current implementation of this feature, it needs to be reworked. I suggest to disable it in this case.

**3) And the most "interesting" one. Issue caused by missing "G92 E0" command in "before layer change script". This command should be used in the combination with relative extrusion mode. I will add it to our Ender-3 profile.**
[image]
[image]

Fixed 3MF (repaired models, added G92 E0, disabled avoid crossing perimeters).
[githubmissileRacks_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/5387635/githubmissileRacks_fixed.zip)

nerobro, 2020-10-15:
Thank you, that's good feedback.

This asks the question: why is it fine when I do it with less than say.. 40 models on the build plate?

rtyr, 2020-10-15:
2) less calculations of travel paths
3) less objects, less accumulation of error (which is the reason why G92 E0 should be used, it will reset extruder position after each layer)

nerobro, 2020-10-15:
I'd expect the accumulation of paths to be relatively linear, not... having a hard limit where it suddenly .. cracks under the pressure?  And sending the head home after each layer seems kinda an odd fix.  Is this a way of purging the slicers "where the head is" on each layer change?

rtyr, 2020-10-15:
The firmware internally maintains an absolute counter even if it is addressed relatively in the g-code. Same goes for firmware simulator in the PrusaSlicer. This G92 E0 resets the internal counter, thus preventing the accumulation of errors.

nerobro, 2020-10-15:
That makes sense.  I take it, this is something of an edge case.  ;-)  hah.

rtyr, 2020-10-15:
I would not expect such big impact to be honest.

bubnikv, 2020-10-16:
We have indeed seen such effect not only in slicer, but also with our Prusa printers running an Arduino based Marlin firmware with 32bit float accumulators for the extrusion. It was surprising to us indeed.

bubnikv, 2020-10-16:
> That makes sense. I take it, this is something of an edge case. ;-) hah.

It is not an edge case. It is a real problem. We have usually seen this with a full print bed print after the first third or half of the layers being printed.

Resolution of mantissa is just 24 bit, that is roughly 17 meters of filament if the filament position is measured in micrometers.
https://en.wikipedia.org/wiki/Single-precision_floating-point_format

bubnikv, 2020-12-11:
I think the detail loss issue due to the extruder axis rounding has been explained.

"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

closed 2020-12-11 by bubnikv