prusa3d/PrusaSlicer #4887: Infinite loop when run from command line

gilesp1729, 2020-10-17:
PrusaSlicer 2.2.0, Windows 10. Command line running.

Specifying fill density as percentage: "--fill-density 30%" works OK.
But leaving out the "%" sends program into an infinite loop at high CPU.

Working example (any STL will do)

D:\dev\LoftyCAD-MG>"c:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --fill-density 30% -g 2shapes.STL
System OpenGL library released
Loading MESA OpenGL library: c:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
88 => Generating skirt
90 => Exporting G-code to 2shapes.gcode
Slicing result exported to 2shapes.gcode

Failing example (only difference is no "%" on the fill density)

D:\dev\LoftyCAD-MG>"c:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --fill-density 30 -g 2shapes.STL
System OpenGL library released
Loading MESA OpenGL library: c:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill

(hangs here until killed)

bubnikv, 2020-12-07:
PrusaSlicer 2.3.0-beta2 gives an error:

```
C:\data\src\PrusaSlicer\build\src\Debug\prusa-slicer-console.exe --fill-density 30 -g D:\3dprint\20mmbox.stl
error: Value out of range: fill_density
```

```
C:\data\src\PrusaSlicer\build\src\Debug\prusa-slicer-console.exe --fill-density 1 -g D:\3dprint\20mmbox.stl
error: The selected fill pattern is not supposed to work at 100% density
```

```
C:\data\src\PrusaSlicer\build\src\Debug\prusa-slicer-console.exe --fill-density 0.5 -g D:\3dprint\20mmbox.stl
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
88 => Generating skirt
90 => Exporting G-code to D:\3dprint\20mmbox.gcode
Slicing result exported to D:\3dprint\20mmbox.gcode
```

bubnikv, 2020-12-07:
Works in PrusaSlicer 2.3.0-beta2.

closed 2020-12-07 by bubnikv