prusa3d/PrusaSlicer #4888: 3Dconnexion Spacemouse Jerky Movement only in Sliced View

cosmith71, 2020-10-17:
PrusaSlicer 2.3.0-alpha1
Windows 10 Version 10.0.19041 Build 19041
3Dconnexion driver 10.7.3248 (latest at the time of this writing)

Spacemouse works perfectly on the plater screen.  However, the sliced preview screen has very jerky movement and is not smooth at all.  This is the case on .stl files or with no .stl files loaded (blank plate).

Regular mouse control works as expected.

I have tried re-installing the spacemouse drivers.  No change.

Thanks.

bubnikv, 2020-10-17:
I am afraid the jerky movement is due to the complexity of the print path preview. The frame rate is simply low.

cosmith71, 2020-10-17:
Thanks for the reply, Vojtěch.  

Just to clarify, it works fine with a regular mouse, and it's jerky with the spacemouse even with no object loaded.  The spacemouse will act differently than a regular mouse?

Area5142, 2020-10-17:
I see the same on Linux in "3D editor view" and "preview" uneven motion when using 3DConnexion mouse, independent of model size. Zoom is worst.

No problem with regular mouse.

cosmith71, 2020-10-17:
I might add that this is a new issue for me with this version.  It works fine in 2.2.0.

foreachthing, 2020-10-17:
> Spacemouse works perfectly on the plater screen. However, the sliced preview screen has very jerky movement and is not smooth at all. This is the case on .stl files or with no .stl files loaded (blank plate).

Move your normal mouse at the same time you use the 3DX device. It works with my SpacePilot.

cosmith71, 2020-10-18:
> Move your normal mouse at the same time you use the 3DX device. It works with my SpacePilot.

Oddly enough, this works with my Spacemouse Wireless as well.  It's not really a feasible workaround, but might be of interest to the programmers.

jweaver100, 2020-10-18:
Same here.. But as you say, moving the regular mouse around whilst you use the 3DMouse fixes it.. So it can't be a frame rate issue.

I have just raise another topic on the same subject.. In 2.3.0 I can't reconfigure the 3DMouse so its too slow to use.. The old CTL+M dialogue has gone... And changing it in the 3DxWare drivers does nothign for me.. Problem is, I have to use the old drivers (10) because I have an old SpacePilot...

Anyone else had problems?

foreachthing, 2020-10-19:
https://github.com/prusa3d/PrusaSlicer/issues/4888 fixes it for me!
Thanks a lot, @enricoturri1966 !!!

cosmith71, 2020-10-22:
Seems to have fixed it for me as well.

Thanks!

bubnikv, 2020-10-24:
Will be fixed with PrusaSlicer 2.3.0-alpha2. Closing.

closed 2020-10-24 by bubnikv