prusa3d/PrusaSlicer #4889: PrusaSlicer 2.3.0-alpha1 does not start on macOS Mojave

mathiasvr, 2020-10-17:
### Version
2.3.0-alpha1

### Operating system type + version
macOS Mojave 10.14.6

### Behavior
* _Describe the problem_
The PrusaSlicer app cannot be started.

* _Steps needed to reproduce the problem_
Click to open the app or start from terminal.
* _Expected Results_
The program starts.
* _Actual Results_
The program does not start.
Running from terminal with outputs the following logs:
```
/Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --loglevel 5
[2020-10-17 18:13:04.897170] [0x0000000115eab5c0] [debug]   full path: 1840334539247780131
[2020-10-17 18:13:04.897436] [0x0000000115eab5c0] [debug]   single instance: undefined. other params: /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer;--loglevel;5
[2020-10-17 18:13:04.897448] [0x0000000115eab5c0] [debug]   full lock path: /Users/mathias/Library/Application Support/PrusaSlicer-alpha/cache/1840334539247780131.lock
[2020-10-17 18:13:04.898105] [0x0000000115eab5c0] [info]    instance check: Another instance found. This instance will terminate.
```
Seems like it detects another instance running, but this is not the case.

sgomes, 2020-10-17:
I'm running Catalina and seeing the same issue. 

I did find a workaround, though: running `/Applications/PrusaSlicer\ Alpha.app/Contents/MacOS/PrusaSlicer --no-single-instance` forcibly disables single instance mode, so the application loads.

Perhaps the instance check is broken on Mojave?

mathiasvr, 2020-10-17:
Thanks @sgomes, running with `--no-single-instance` works for me as well.

mik-at, 2020-10-17:
Same Problem on High Sierra, running with `--no-single-instance` helped. Thanks!

JohnOCFII, 2020-10-17:
I didn't have that issue on first-launch, but did on subsequent launches until I tried launching from the command line with `no-single-instance` as an argument.  After that, I've been able to launch successfully by double-clicking the app icon.  Could this be related to previous alpha testing and/or the state of the /Users/<myuser>/Library/Application Support/PrusaSlicer-alpha/ directory?

macOS Catalina 10.15.7 here.

mathiasvr, 2020-10-17:
@JohnOCFII You are right I can also open the app by double-clicking after having used `no-single-instance` once.

lukasmatena, 2020-10-17:
Thanks guys, We will fix it. It is indeed connected to previous application datadir status.

jensus11, 2020-10-18:
> I'm running Catalina and seeing the same issue.
> 
> I did find a workaround, though: running `/Applications/PrusaSlicer\ Alpha.app/Contents/MacOS/PrusaSlicer --no-single-instance` forcibly disables single instance mode, so the application loads.
> 
> Perhaps the instance check is broken on Mac OS?
hello how do i have to do this? do I have to enter this in the terminal?

lukasmatena, 2020-10-19:
Should be fixed in master with a47e94e.

@JohnOCFII Could you please test with current master (you are able to compile it, right?)? I'm quite puzzled by what you're saying. The problem was that if the datadir did not exist (people who did not run any alpha builds yet), it thought that another instance was running. You should not have suffered by this (and you didn't), but I have no clue why would it break for you later. This may be another problem.
If you could test the scenario with no datadir as well, that would be awesome (just rename it for a minute).

malcinator, 2020-10-19:
I have seen the same issue and from what I have found if the directory $HOME/Library/Application Support/PrusaSlicer-alpha is missing the app won't start.

JohnOCFII, 2020-10-19:
I'm not currently set up to compile it, but have in the past, and could again in the future.  Based on the notes above I _think_ what happened is that on first launch, which worked fine, I noted it pulled in all the old profiles, etc., from the last time I ran an alpha for 2.2.x.  I decided I wanted to truly start fresh, so I exited PrusaSlicer, than manually deleted everything in the $HOME/Library/Application Support/PrusaSlicer-alpha directory (or I might have taken the directory itself -- don't recall exactly).  Then I started to have the issue until I did the manual command line launch with the `no-single instance` argument.

lukasmatena, 2020-10-19:
@JohnOCFII Thanks for your response. If you think you may have obliterated the directory yourself, it would explain it. If you cannot compile now, that's fine. It makes sense now.

bubnikv, 2020-10-24:
Will be fixed with PrusaSlicer 2.3.0-alpha2. Closing.

closed 2020-10-24 by bubnikv