prusa3d/PrusaSlicer #4890: Slicer skips top layer

finewhin, 2020-10-17:
### Version
2.2.0+

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macOS Catalina

### Behavior
The Slicer seems to ignore the Top Layer, they come out heavily under extruded, if I look into Gcode viewer it says the model is 19.8mm and the slicer says it is 20mm so this seems to be the issue.

rtyr, 2020-10-17:
I am afraid I don't understand your issue, please provide your 3mf project file. I can see from the gcode that you have set 0 top layers and the last layer is printed at Z20, not 19.8.

finewhin, 2020-10-18:
I'm sorry I uploaded the wrong file. To me it seems like PrusaSlicer ignores the top layer, the exported file has a height of 19.8mm although it should be 20mm high. Also the top layer is heavily under extruded unless I over extrude the rest of the print. The layers under the last top layer are perfectly fine.
[Form-Kubus.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5397759/Form-Kubus.3mf.zip)

rtyr, 2020-10-18:
I don't see any problem in the gcode. The last layer is printed at 20mm height as expected. 

[image]

Printed gcode:
[image]

Your issue is most likely HW related.

finewhin, 2020-10-22:
Did you measure the printed cube? I sent the sliced gcode to someone else who confirmed that the print came out one layer to small. So the under extrusions seemed to be HW related since I fixed that. But the prints keep getting out 1 layer to small 19.8 instead of 20mm, and I'm pretty sure this isn't HW related. If I look into the Gcode everything seems fine, it shows the G command for 20mm height. But every printer/gcode analyzer seems to ignore that.

rtyr, 2020-10-22:
Yes, the real Z height is ok as expected. You can see, that the last layer is printed at Z20.

The preview is shown correctly in every gcode viewer I tried, including gcode analyzer you mentioned. The model info in the gcode analyzer shows 19.8, but it is just some sort of bounding box, which is most likely not reliable (it started to show 20mm when I added intro line to your start g-code...)

[image]

rtyr, 2020-10-30:
Closing. There is no indication that the issue is caused by the PrusaSlicer.

closed 2020-10-30 by rtyr