prusa3d/PrusaSlicer #4892: 3Dconnexion settings (command + m) are missing

dahuafschmied, 2020-10-17:
### Version
2.3.0 Alpha1

### Operating system type + version
MacOS 10.13.6

### 3D printer brand / version + firmware version (if known)
Mini 4.2.1

### Behavior
3dconnexion settings menu (command + m) is missing,
In Version 2.2.0 with Command + m the menu is displayed

in adition the speed default values are way to low. i have to set all 3 of them to 10 to get the same speed as in other software.

Workaround:
edit the PrusaSlicer.ini file

dahuafschmied, 2020-10-17:
same in new g-code viewer

bubnikv, 2020-10-17:
We dropped the built in configuration dialog as you should be able to do the same configuration and more in the 3DConnexion driver.

dahuafschmied, 2020-10-17:
ok. sounds logic.
then please set the speeds to 10 to have the same possibilities as in other software.
if the speed in prusa slicer is set to 1 and the speed settings in the driver to maximum i can not even get the default speed of all other programs.

foreachthing, 2020-10-17:
Same here! 
Speed settings too (s)low.

jweaver100, 2020-10-18:
For me, the 3DxWare config is not even working.. It has no effect, so the speed of my movement is too slow to use..

I want the old functionality back as it was PERFECT for me..  SInce this is thing I only need to setup once, is there a way to force my 3DMouse speeds to be faster?

dahuafschmied, 2020-10-18:
My suggestion would be to set the default values to 10 and leave the configuration dialog disabled.
Because no one would need it when the speeds are correct.

jweaver100, 2020-10-18:
Agreed.. Happy to manually set this up and leave it.. But how?? I can'ty find any way to change the speed...

Is there a file I can edit? Wait.. Just saw the update in the original post.. "prusaslicer.ini".. Great.. I have that.. But what params to you edit?

Boby71, 2020-10-19:
My "Space Mouse compact" works flawlessly on the object platter, but does **NOT** work on the preview platter...

jweaver100, 2020-10-19:
> My "Space Mouse compact" works flawlessly on the object platter, but does **NOT** work on the preview platter...

Move the normal mouse at the same time and it should work in the preview mode.. fix is inbound apparantly.

foreachthing, 2020-10-19:
> > My "Space Mouse compact" works flawlessly on the object platter, but does **NOT** work on the preview platter...
> 
> Move the normal mouse at the same time and it should work in the preview mode.. fix is inbound apparantly.

This commit https://github.com/prusa3d/PrusaSlicer/commit/5d3d9d9b9250db8474659dd144524e1aa9b35406 fixed it for me :v:

dahuafschmied, 2020-10-22:
> > > My "Space Mouse compact" works flawlessly on the object platter, but does **NOT** work on the preview platter...
> > 
> > 
> > Move the normal mouse at the same time and it should work in the preview mode.. fix is inbound apparantly.
> 
> This commit [5d3d9d9](https://github.com/prusa3d/PrusaSlicer/commit/5d3d9d9b9250db8474659dd144524e1aa9b35406) fixed it for me ✌️

But does NOT fix hijacked this Ticket!

cosmith71, 2020-10-22:
I'm having the same issue, even after editing the prusaslicer.ini file.  The 3Dconnextion 3DXware does not affect the speed in any way, nor does editing the values in the prusaslicer.ini file.

foreachthing, 2020-10-22:
> But does NOT fix this hijacked Ticket!

You're absolutely right!!!
CTRL+M has been removed from PS and been replaced with the "normal" way of integrating such devices.
And the speed settings, on my computer, is not saved either and reeeeeeally slow!!!  :zzzzz:

dahuafschmied, 2020-10-24:
still not fixed in alpha 2!
it is just a simple default value change. why is this not done??

foreachthing, 2020-10-24:
I, too, wish they would revert to the perfectly working "CTRL+M" version.

jweaver100, 2020-10-24:
Going back to old configurator gets my vote too... i had zero issue with the old implimentation.

Perhaps it could be a configurable option to have the new or old method?

bubnikv, 2020-10-25:
I see the discussion is getting hot and the pitchforks are getting raised.

We are trying hard to do the right thing to integrate 3DConnexion into PrusaSlicer according to 3DConnexion recommendations. We are getting some support from 3DConnexion including their people testing PrusaSlicer for the right default values for the speeds. 3DConnexion normally provides libraries for integration into closed source projects, however as PrusaSlicer is open source, we had to find a way to integrate without the 3DConnexion libraries.

We are trying our best, but we are not quite sure whether the issue is on our side or on the side of 3DConnexion. On Windows we seem to be doing good. On OSX the 3DConnexion messages are supposed to be routed through the 3DConnexion driver, thus the 3DConnexion driver should be able to adjust the speeds. If that does not work as expected, we cannot know whether the issue is on our side, whether the 3DConnexion driver is buggy or whether the devices that you guys report are now too slow in PrusaSlicer are simply not supported by the 3DConnexion driver.

Frankly I am getting hard time to orient myself in the issues reported. I suppose the best we can do is to adjust the default speeds for the devices we have access to and re-enable the Ctrl-M dialog for those devices that are getting incorrectly scaled values and hope for the best. We will most likely do that for the next alpha. We will also consult with 3DConnexion, however they are not exactly quick to communicate.

Knowing up front how much development time the 3DConnexion integration would take, as a project manager I would most likely refuse to implement this feature.

jweaver100, 2020-10-25:
@bubnikv From my point of view there is no need to justify your actions.. I know you are trying to do the best thing, and its not difficult to see how difficult it is to work with 3DConnexion.. The way they "obsolute" there products is nothing short of criminal.. So trying to build support for some of their products is likely to be impossible.

All i am saying (And I believe others too) are that for people who liked the old implimentation of the 3DMouse, bring it back.. You spent so much time developing it, it seems a shame not to use it when it worked for many people.

The fact that you are doing this in a future realise is fine by me and I couldn't be happier.

Just keep doing what you are doing.. 

Cheers

Jon

dahuafschmied, 2020-10-26:
@bubnikv 
Thank you so much for implementing these devices. I appreciate it a lot!
I see here two different things accumulated in this ticket.

1. Mouse is to slow (MacOS) with supported devices (SpaceNavigator, SpaceMouse, ...).
Settings can be changed in 3DxWare. But are still too slow.
(that is my case and the original intend of this ticket)

2. Mouse is to slow with NOT supported devices (SpacePilot,...).
Settings can NOT be changed in 3DxWare and/or INI file.

The Problem i see is that a lot of people using the unsupported devices because of there low price. If you reenable the cmd+m Settings the complains will disappear and you have time to find the speed problem (for a final solution i see this necessary).

bubnikv, 2020-10-26:
What exactly are the non-supported devices? Are they 3DConnexion products
that 3DConnexion obsoleted? Or are they clones of 3DConnexion devices?

What surprises me is that these "non-supported" devices still seem to
register at the 3DConnexion driver and the 3DConnexion driver produces
movement messages for them. At least this is my understanding of it.

dahuafschmied, 2020-10-26:
Here you can find the list of supported devices:
[Driver Release Notes](https://17osrg460vnh32txvs1x75a9-wpengine.netdna-ssl.com/us/wp-content/uploads/sites/27/2020/08/Release_Notes-Mac-20200804.pdf)

and a list of [discontinued devices](https://3dconnexion.com/us/support/faq/3dconnexion-discontinued-devices/)

As i see the people with unsupported devices just install old drivers (and have compatibility problems now)

jweaver100, 2020-10-26:
@bubnikv In my case its a "SpacePilot".. A perfectly useable product and an official 3DConnexion device.. But they remove support for their products after a few years.

so mine still works, but only using old drivers and this means that a lot of new apps (ie Sketchup) doesn't support it.

Its 3DConnexions business model to keep dicontinuing their devices.. I guess if you were a "pro" user, you wouldn't mind upgrading every few years.. But for people like me, I have an old 3DMouse and am happy to use old but unsupported drivers.

jweaver100, 2020-10-27:
@bubnikv I just noticed there is a commit to bring back the old CTL+M functionality, but I noticed its for MAC.

When came back to look at this thread I realised it was raised against MACOS, where as my issue is in Windows..

Is it also possible to bring back the old functionality to Windows too?

dahuafschmied, 2020-10-28:
Hey guys
i have done some investigations:

- At Windows there is the same problem. Mouse is to slow.
- No settings in INI file to fix it.
- Changing the speed in 3Dconnexion Driver changes the speed but not enough.

Summary: 
Windows has the same problems as MacOS. or even worser. (no ini file fix)

Recommendation:
Bring back cmd+m for Windows too.

enricoturri1966, 2020-11-03:
@dahuafschmied, @foreachthing
Could you please test the following experimental build where the settings dialog (ctrl+m) for 3DConnexion devices on Windows has been restored, and let me know if this solution would work for you ?

https://www.dropbox.com/s/2iqmt4l7ysb8nf7/PrusaSlicer-2.3.0-alpha2%2B105-win64-g54d683455-202011030753.zip?dl=0

jweaver100, 2020-11-03:
@enricoturri1966 I have tested and the CTL+M configurator is back.. And the values are set to 1.0

So it all looked good, but I changed the values and it has now effect on the speed of rotation... So we are making good progress, but its not working yet.

For the record, I am one of the people who had ZERO issues with SpaceMouse support in 2.2.0 so am happy to help test this.

foreachthing, 2020-11-03:
Yeah, I'm with @jweaver100.
* Speeds to slow
* Dialog is back, has no effect and is twitchy
* Settings are not saved

Happy to do some more testing!

tg73, 2020-11-03:
Same for me as @foreachthing. I'm also happy to test. Using SpaceMouse Pro (wired) with Win10 x64.

dahuafschmied, 2020-11-04:
in alpha3 command+m sill not working on mac os
no dialog is shown

enricoturri1966, 2020-11-06:
@dahuafschmied, @foreachthing, @jweaver100
Could you please test if this new experimental build fixes the issues with settings dialog (ctrl+m) for 3DConnexion devices on Windows ?

https://www.dropbox.com/s/vvp6u4ebab1tb4n/PrusaSlicer-2.3.0-alpha3%2B22-win64-g277376ee0-202011061044.zip?dl=0

jweaver100, 2020-11-06:
@enricoturri1966 
The joys of "working from home" means I can test this in real time.. And I have good news....

IT WORKS!! Just like it used too in 2.2.0

I will do more playing, but my basic tests worked.

1. Lauched PS - 3D Mouse Slow
2. CTL+M - GUI Appeared (all set to 1.0)
3. Increased them to 9+
4. 3D Mouse now fast
5. CLosed PS
6. Lauched PS
7. Setting remain...

So from what I can see, its all working normally again..

dahuafschmied, 2020-11-06:
@enricoturri1966 
Sorry! No Windows available at the moment. We are a pure mac company. Next week perhaps i have access to a Windows notebook.

cosmith71, 2020-11-06:
Not that you asked me, but that works for me as well and as jweaver100 describes.  Windows 10 and Spacemouse wireless.

foreachthing, 2020-11-06:
Sweet!!!!! :heart_eyes:
Just like @jweaver100 described: everything works as expected! :godmode:

tg73, 2020-11-06:
@enricoturri1966 As per @jweaver100 , your latest test build works well for me Win 10 x64, SpaceMouse Pro (wired), a value of 7.5 feels good to me.

dahuafschmied, 2020-11-09:
@enricoturri1966 
cool that windows is ok now. but my (this) ticket is about MacOS and there is still no command+m dialog in alpha3.
just to bring this up again. ;-)

dahuafschmied, 2020-11-21:
@enricoturri1966 still no command + m on macos with alpha4

jweaver100, 2020-11-24:
> 
> 
> @enricoturri1966 still no command + m on macos with alpha4

Nor Windows.. Which was quite a surprise after the successful "test" version based on Alpha3.

bubnikv, 2020-11-28:
We have brought the Ctrl-M dialog back in PrusaSlicer 2.3.0-beta1. You have to enable "Legacy 3DConnexion" in the preferences dialog first.
Closing.

closed 2020-11-28 by bubnikv