prusa3d/PrusaSlicer #4893: Feature Request: Printer configuration include function

TheAndr0id, 2020-10-17:
### Version
2.2.0

### Operating system type + version
Windows 7 / x64
Linux x86_64

### 3D printer brand / version + firmware version (if known)
Ender 3 V2

### Behavior
Printer configuration is a monolithic file containing all configurations for a given vendor.   If a change is needed then the entire file needs to be updated.  In my case: [resources/profiles/Creality.ini](https://github.com/prusa3d/PrusaSlicer/blob/master/resources/profiles/Creality.ini)

I have an Ender 3 V2, and while it is VERY close to the Ender 3 there are small changes due to the new motherboard/steppers/bed.  Adding a new configuration to PrusaSlicer was complicated and every time a new configuration is released I have to do an update to my configuration file to support the update.

Could we please get an **include** function added to the config files?  This would allow for modular configurations and only updates to files which need to be updated.  If the include function allowed for directories to be specified this would also allow for simpler directory trees as well.  For instance the Creality.ini file could basically be:

`include = Creality`

which would then look for all ini files under resources/profiles/Creality and enumerate them with POSIX file ordering.

That way adding a new printer would involve adding a NEW file not editing a long complicated rambling configuration.  This would also simplify debugging and testing of new printer configurations as other configurations would not be impacted by new additions.

_Is this a new feature request?_
Yes

bubnikv, 2020-10-17:
We could make the configuration system more complex than it is, but wouldn't it be better, if you contribute your profiles to us, so we can integrate them and share them with the community?

TheAndr0id, 2020-10-17:
I'm not willing to provide support for my configuration.  I've taken values from 8 different profiles and combined them in ways I can't guarantee will work.  I'm still tweaking and modifying it almost every print.  Sometimes it does a good job, sometimes I get goop.  :D

I've also had very bad experiences in the past contributing to projects.  After getting burned and frustrated I just developed a policy of "not again".    Please don't take this as being selfish, I just don't need the headaches.

An include function doesn't need to be that complicated.  it's just opening multiple file streams and feeding them into the same place you are feeding the current configuration file.  Many, many other software packages do this.  I do this in shell scripting all the time...  (I'm a Unix System Administrator)

Logic:
- Read main config file
- Hit an include
     - Is it a directory?
           - Loop files
     - Read file
- continue with main file

Wash, rinse, repeat.  I bet there's libraries out there (probably something in the Boost libs) that will do this for you.

I can see a recursive solution here.  Yes, it might not be super memory friendly, but it's a one time call on startup.  The end result is going to be the same with the same amount of data being stored.

I've been experimenting with the SuperSlicer fork, and their Creality.ini file is almost 40% larger due to the support for the CR-10s and extra nozzle sizes they've added.  It's also very difficult to follow,  ~700 lines of config file is not a good thing.  Ask the guys who maintained sendmail how long complex configurations worked out for them.  Postfix now owns the mail service with it's easy to read configuration under 50 lines per file.  Sometimes adding some processing complexity saves everyone masses of overhead in the long run.  (how can you tell I'm a DevOps guy? :) )

I'm just sayin' ...

closed 2026-09-29 by prusaslicer-bot[bot]