prusa3d/PrusaSlicer #4895: Missing Prusament PC Blend profile from 2.3.0-alpha1

4nthonylin, 2020-10-17:
### Version
Version 2.3.0-alpha1+win64

### Operating system type + version
Windows 10 Build 19042
### 3D printer brand / version + firmware version (if known)
Prusa Mini 4.2.1

### Behavior
Missing Prusament PC Blend Filament Configuration
Current - 2.3.0-alpha1
[image]

Expected - 2.2.0
[image]

rtyr, 2020-10-17:
Restart the PrusaSlicer and update your configuration. 

[image]

You should have version 1.1.10.

[image]

rtyr, 2020-10-18:
Can we close the issue?

4nthonylin, 2020-10-19:
Yes that worked, thanks! Must've not seen the configuration update.

closed 2020-10-19 by 4nthonylin

devrosx, 2020-10-22:
for some reason PC-blend profile is missing on Mac and czech language, when i switch to english it appears in menu... (filament profile selection)

rtyr, 2020-10-24:
@devrosx Can you please send a screenshot of the filament wizard (czech language)?

devrosx, 2020-10-25:
tried to replicate with latest alpha version and problem is gone now.... sorry :)