prusa3d/PrusaSlicer #4896: Supports generated where I do not want them

tuxmobil, 2020-10-18:
### Version
2.3.0-alpha1

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
mk2.5s

### Behavior
I painted supports where I want them but PrusaSlicer also generated supports for (part of) the lower screw hole.
I then painted that screw hole to not have any supports, but that didn't help.

I also had this issue with 2.20 custom supports so it may not be related to the painting feature at all.

[image]

#### Project File (.3MF) where problem occurs

[Painted Supports.zip](https://github.com/prusa3d/PrusaSlicer/files/5397290/Painted.Supports.zip)

lukasmatena, 2020-10-18:
The supports are not generated for the lower screw hole. They are generated for the top overhanging part. If you look into the hole, you'll find no supports there. Set supports "Everywhere" and you'll see the difference.

This is not caused by the supports painter, but by the way how PrusaSlicer generates supports. Not even the good old support blocker helps here.

tuxmobil, 2020-10-18:
Thanks for looking into it!I noticed that the support does not reach in to the hole completely. However, as it is not needed here and cumbersome to extract from a tiny screw hole, I would appreciate if PrusaSlicer would either not generate them or let me remove them conditionally.I noticed that support blocker didn't help me here, hence the issue.

bubnikv, 2021-09-22:
solved with snug supports in PrusaSlicer 2.4.0-alpha1.

closed 2021-09-22 by bubnikv