prusa3d/PrusaSlicer #4898: [FEATURE REQUEST] PrusaSlicer Cloud Settings - Sync Settings between PCs

weltermarcel, 2020-10-18:
### Version
2.2.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S

### FEATURE REQUEST

Hi,

I use PrusaSlicer on several computers and each time I make tweaks on my print, filament or printer setting profiles, I have to export a ConfigBundle and import them on the other computers manually. I store the bundle file on a gdrive folder to make the syncing a little easier but if for any reason I forget to update the profiles on any of the machines with the latest profile settings and I then tweak some other things, then my new exported bundle contains a mix of old and new settings (depending on what I just changed / changed last time on the other computer)...

Now, another way to do this would be to automatically sync the .ini files on "C:\Users\USER\AppData\Roaming\PrusaSlicer\" but the problem is that not all of the computers I use are on the same network and also not all of them run the same Prusa Slicer version all the time and I am not sure about cross-compatibility.

So a nice solution to this would be to be able to **log into your account in PrusaSlicer and have all the settings stored on the cloud** and getting to chose which settings should sync and which others not (like you get to do using the dropbox or gdrive apps). Also each time you open PrusaSlicer you could get a warning message letting you know there are new profiles available?

Maybe this is already on the pipeline or someone knows a workaround to this "issue"?

Marcel

KD7LRJ, 2020-10-18:
Here's what I do in my Windows shortcut settings for starting Prusa Slicer (include the quotes as shown):

- Target: "C:\Program Files\Prusa3D\PrusaSlicer_2.3.0Alpha0_05.July.2020\prusa-slicer.exe" -datadir "G:\My Drive\3D Prints\PrusaSlicer\PreRelease_Settings"
- Start In: "G:\My Drive\3D Prints\PrusaSlicer\PreRelease_Settings"

I do need to restart the app to get it to reload the settings, but this has been working for me. I do not use multiple platforms, so YMMV.

   ...Doug

renarsvilnis, 2020-12-24:
@KD7LRJ What folder does the PrusaSlicer for Windows contain? Wondering if it similar to what I have in MacOs in `/Users/renarsvilnis/Library/Application Support/PrusaSlicer` (see screenshot).

I'm currently looking into syncing settings between two Macs, have no issue creating symlinks from google drive as I do it already for a couple of dev-tools.

[image]

KD7LRJ, 2020-12-27:
> @KD7LRJ What folder does the PrusaSlicer for Windows contain?

The only addition in mine is a folder named "physical_printer," otherwise my list is identical to yours.

I should mention that using that option overrides the default behavior of the pre-release vs. release versions that try to keep their profiles in different locations. I just try to make a copy of the files occasionally, but mostly rely on the automatic snapshots taken during upgrades, etc..

As a side note that could probably be reported separately... The application acts like it's locked up when taking configuration snapshots. I'm guessing that when most people run that function on local drives, it's really quick and not noticeable, but when using a remote drive, it's a bit disconcerting for the 67 seconds that it takes (I just timed it) to start responding again. And before anyone asks about the speed of my internet connection...

[image]

BraveSolo, 2022-05-29:
Why is this not baked into the Slicer?  Its SO OBVIOUS!  I run prusa slicer on 2 machines, and need to sync them_

acekrystal, 2022-10-12:
I really need this feature too!

I wished Prusa would add better support for centralized configuration files... Since a few updates it has become harder and harder to keep using the "--datadir" way :
- Every update you need to go and find all shortcuts and edit this --datadir back in.
- If you open prusa-slicer related files directly sliver does not get opened through a modded shortcut and does not load in your config files
- Now these day's there are 2 shortcuts "Configure Automatic Updates" and "PrusaSlicer 2.5.0", I don't know if certain function stop working by only updating and making new shortcuts directly to "PrusaSlicer 2.5.0" file.
- Following up on the line above and [this github #7567 issues](https://github.com/prusa3d/PrusaSlicer/issues/7567), clearly feathers do fail and this is the first one I visually get feedback from that it fails. (Send System Info screen keeps opening on startup)

I'm using the --datadir to save my config files to a nextcloud synced folder, so I can use if over multiple PC's and even on laptop on remote locations to do my work. But it is just far from optimal. I even had one time the default config's overwrite my synced config files, losing all my configurations while at work by a customer under time pressure! I managed to get the files back by reverting backups and am now making backups before every update... but there is just so fragile using the --datadir way. 

**Please add a proper way to point to a synced profile location!**

DragonFinlay, 2022-10-12:
While I fully agree that the drawbacks of the datadir method are annoying and it would be nice to have a "proper" way to do this, think about how many people actually need something _need_ this feature.

PrusaSlicer is mostly used by owners of Prusa printers, which are especially interesting if you're new to the 3D printing scene. I imagine the majority of our users would never make use of this feature at all, it's mostly "power-users" like us with more than one PC for slicing, or PCs in different locations.

I'll check with the team what is planned in this regard and how it can be implemented. Thanks for your comments!

MichaelKetchel, 2022-11-14:
I can attest to this being a royal pain in the butt. I have 2-3 computers that I do design and printing from; My very powerful desktop with multiple monitors for primary design and slicing, which is in a completely different end of the house, and either my Surface or laptop for tweaking and slicing from the print room. Additionally, I work in both Windows and Linux (because Fusion is windows only....sadness). I use Syncthing to try to selectively sync files, because the contents  of the data directory are too tightly coupled with both the version of Prusaslicer as well as the platform, so syncing sometimes breaks things badly. I'd love to just store the datadir on my NAS, but for the reasons above, bad juju. But every time I come back, I'm left wondering where my definitive, most recent filament and slicer profiles are, because the way things save is weird if I forget to close Prusaslicer on another machine or one updates before the other. This is immensely frustrating, and knowing this headache awaits me often discourages from even starting printing projects, as mine often involve some tweaky filaments like TPU.

I don't care if Prusaslicer necessarily implements an internal sync, but I'd love to at least see it store data files in a more sync-friendly way; keeping the OS and local settings/recent files etc separate from the important stuff like printer and filament profiles, and perhaps being mildly aware of the version of those files so that when you update one instance, the others don't puke because the config is in the new format, but rather prompt for an update. The user can then use their favorite file sync system.

I think separate, dedicated printing areas are more common than you'd think, and are becoming more so as people become more aware of the issues surrounding particulate emissions from FDM printers, or fumes from filaments like ABS. Most people I know have their printers in their own room or closet now.

rangersnuggles, 2023-06-09:
I have: 
A desktop Mac in my home office
A desktop Mac in my office office
A Mac Laptop
A workshop PC running Windows

A LulzBot Mini in my office
A Prusa Mini, An Ender 3 S1, A Voron 2.4 and a Voron 0.2 en route.

I would *really* love a good way to keep Prusa Slicer settings consistent and synced between everything.

tr1pp4, 2023-07-25:
would love to see this too.

Actually i run Prusa Slicer on one machine only, but rectently my SSD died and of course i do not backup all my settings frequently. It's 2023, we should spend time for other things than manual backups 

Prusa could at least add a feature to automatically save the settings to a specified path, it's really not rocket science. 
As a workaround i will setup a "mklink" to OneDrive, same a i do for other tools.

anghelos, 2024-01-26:
I've made these scripts which kind of solve this for my use case, maybe it could be useful to you too: [anghelos/prusaslicer-config-sync](https://github.com/anghelos/prusaslicer-config-sync). I've also attempted a version for Mac/Linux, but haven't had the opportunity to thoroughly test it.

jfxg, 2024-08-18:
Another +1 for this one. I have a Prusa MK4 and a Bambu X1C. My wife recently started printing and we setup both slicers on her laptop. In Bambu Studio, she installed it, logged into my account, and clicked "Yes" when it asked to sync my configurations. She was done and ready to go. Prusa Slicer was far more work getting the configuration sent over to her laptop, and we know its going to be a hassle if we ever add new custom profiles.

This is also why I only ever use one computer for my Prusa, but I often find myself printing to the Bambu from multiple computers and even my phone for simple models.

PLEASE add this feature.

BraveSolo, 2024-08-20:
I Print from Octo Print and Octo Everywhere which works great from around
the world even.  Just not slicing profiles, lol.I have seen a new feature
with credentials in prusa slicer, so maybe synchronization is next.

Sincerely,

Andy Boles
503.440.3056

stritti, 2024-09-12:
I currently lost all my filament settings because my SSD was corrupted. Would be really nice, if there would be a feature to store config files at a cloud directory to prevent such problems or to share through several computers.

I will now try the solution @anghelos created: https://github.com/prusa3d/PrusaSlicer/issues/4898#issuecomment-1912654014

rob-miller, 2024-10-13:
There's an added frustration here that if I could safely run two instances of PS with synchronized configs simultaneously, then I would not have to move from my design-work office to the computer in the garage next to the printer to start the job.  Lovely that I can submit jobs from the slicer, but the slicer next to the printer is the one that stays up to date with what filament is loaded - so the copy on my office PC is mostly useless.

tr1pp4, 2024-10-13:
I'm wondering how Prusa employees synchronize their computers. I only use 2 computers and it's hard enough to keep them synchronized. It must be a pain to synchronize dozens or hundreds of machines.

Why not making it possible to simply keep the relevant files in a custom directory which could be a network path for example. A cloud solution is not even need here in my opinion.

anghelos, 2024-10-13:
> Why not making it possible to simply keep the relevant files in a custom directory which could be a network path for example. A cloud solution is not even need here in my opinion.

I'm fairly certain this is what they propose. You can have PrusaSlicer store its config to a network drive.

rob-miller, 2024-10-14:
> I'm fairly certain this is what they propose. You can have PrusaSlicer store its config to a network drive.

Just need to split the file contents up, as discussed in #8643 and somewhat in the writeup for your scripts on github.  As this is the 'cloud version' request, breaking up the filament settings could allow crowd sourcing more of those settings and even linking out to something like [FilamentColors](https://filamentcolors.xyz/library/).

rob-miller, 2024-11-03:
Just to share, my current workaround is:

- the datadir is in the default place (on a Mac in _~/Library/Application Support/PrusaSlicer_).
- within this, the _filament_, _physical_printer_, _print_ and _printer_ subdirectories are backed up to my cloud solution (for me, this is into a 'sync' directory that is copied to my NAS with SyncThing.  SyncThing on the NAS does version control, other computers sync from here to their own PrusaSlicer DataDir). _shapes_ might be worth adding to this set, but so far I have not used it.
- to actually sync the printer settings (as everything will fall over if you sync PrusaSlicer.ini - see #8643), I have to save a project file (.3mf) and load it into the other PrusaSlicer.

darderik, 2025-02-10:
Even if I doubt at this point that this feature is considered somewhat relevant by contributors, I'd like to state that I had to switch to other slicers (that unfortunately result in lower quality prints) due to the lack of this feature.  In my use case many people use the same printer with different filaments/PCs and the lack of syncing was not viable anymore.

dgarcu, 2025-02-28:
Just a friendly reminder, I use a dual boot system in two machines. Certainly, it's a pain to keep sync manually, or even automatically with the scripts or solutions provided here. 

Now that the Prusa Account is integrated in the slicer itself, maybe is easier to take a step further and store such data within the data account itself, and give the user an option of either grab the settings locally or remotely.

I would really appreciate it, for sure! 😄

redfox-mx, 2026-09-02:
this still is a great feature to add, plis dont close

chking999, 2026-09-02:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

I agree.  Not being able to transfer settings between PCs is a real pain
point.

Christian King

tr1pp4, 2026-09-02:
It's still not planned for PrusaSlicer 3?