prusa3d/PrusaSlicer #4899: 2.3.0 Has broken support for my (SpacePilot) 3DMouse

jweaver100, 2020-10-18:
### Version
2.3.0-alpha1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mk3 and Mini

### Behavior

I was thrilled with the introduction of 3DMouse support in 2.2.0 and I was one of the people with ZERO issues.

But I know a lot of people didn't like how it was implimented and wanted support usign the 3DConnexion drivers/config and I just had a feeling this would be a problem.

Just got to play with 2.3.0-Alpha1 today and the first thing I noticed was that the 3DMouse movement was unbearably slow.. I hit CTL+M to bring up the config and nothing happened.. Then I remembered that config is now down via the 3DConnexcion config tool.. So I loaded it up and started to tweek but it has NO effect on my mouse...

I suspect the reason is that I am uising 3DxWare which is old.. But thats the last release that supports my "SpacePilot"

So I now can't change the speed of movement and 2.3.0 is now a downgrade to 2.2.0.. I would be more than happy with the old config (CTL+M) back.. 

Please do something to help people who do not.. not cannot have the lateast 3DxWare drivers.

jweaver100, 2020-10-18:
It gets worse.. When I first installed 2.3.0 the 3DMouse support was juddery and stuttering.. I then tested with 2.2.0 and it was the same.. And then I went to Sketchup and that was the same too...

So i rebooted and it worked again.. So I thought nothing of it.. When wrote my report above, Mouse Support in 2.3.0 was workign but SLOW and there was no way to change it.

But now its gone to a juddering, stuttering mess again.. And worst still its broken 3DMouse supported in Sketchup too.. So I have to reboot again

So I can't live with this.. The support you have added in 2.3.0 is now breaking my 3DxWare drivers so I hope you can do something.

For me, I would be happy with a config change where you can select 3DxWare support or the old 2.2.0 functionality as that was 100% fine for me!

jweaver100, 2020-10-18:
Just been reading the release log and it talks about 3DConnexion support.. And I am a bit lost.. 

It says that the XML should be created in "c:\Program Files\3Dconnexion\3DxWare\3DxWinCore64\Cfg\prusa-slicer.xml",  but I don't have this.

It goes on to say that the prusa-slicer.xml and prusa-gcodeviewer.xml can be copied into this directory, but where do I get them as they arnt' there by default?

dahuafschmied, 2020-10-18:
duplicate #4892

jweaver100, 2020-10-20:
I am still not sure my issue is a duplicate.. The other issue was fixed by editing the PrusaSlicer.ini.. For me, none of those changes worked.

I just fear I am going to miss out on 3DMouse support in 2.3.0 as I have an old mouse with obsolete drivers and where as 2.2.0 worked perfectly for me, 2.3.0 does not allow me to change the config.

bubnikv, 2020-11-28:
We have brought the Ctrl-M dialog back in PrusaSlicer 2.3.0-beta1. You have to enable "Legacy 3DConnexion" in the preferences dialog first.
Closing.

closed 2020-11-28 by bubnikv