prusa3d/PrusaSlicer #4900: Ironing does not take effect when you change Top Fill Pattern.

jweaver100, 2020-10-18:
Version
2.3.0-alpha1+win64

Operating system type + version
Windows 10

3D printer brand / version + firmware version (if known)
Prusa Mk3 and Mini

Description

If you Tick "Enable Ironing" and then slice, you see the Ironed surface in the Preview.

But if you then change the Top Fill Pattern and leave "Enable Ironing"ticked and reslice, the top layer does not Iron.

You then need to disable Ironing and enable again for Ironing to work.

iLikeDarbo, 2020-10-18:
Version
2.3.0-alpha1+linux-x64
PrusaSlicer-2.3.0-alpha1+linux-x64-202010171312.AppImage

Operating system type + version
openSUSE Tumbleweed 20201014

3D printer brand / version + firmware version (if known)
Creality Ender 3

Description

The same issue as described above happens on the linux version.

Additional case where bug occurs:

1. If you Tick "Enable Ironing" and then slice, you see the Ironed surface in the Preview. Saved gcode includes ironing
2. But if you then change **any parameter under Print Settings** and leave "Enable Ironing"ticked and reslice, the top layer does not Iron. The Plater preview does not showing ironing and the saved gcode does not include ironing
3. Disable and Enable Ironing and the preview will correctly show ironing on and saved gcode includes ironing

bubnikv, 2020-10-22:
Likely a duplicate of #4510

closed 2020-10-22 by bubnikv

ntmaster26, 2020-10-22:
Unsubscribe lol