prusa3d/PrusaSlicer #4905: 2.3.0 issue with OctoPrint + solution

EddyMI3d, 2020-10-18:
### Version
2.3.0-alpha1 +win64

### Operating system type + version
Win 10 
### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2.5S + MMU2S Firmware 3.9.1. + 1.0.6
### Behavior
The actual Start Gcode for single filament use creates problems with OctoPrint (abort with M112 error)

Moving the `Tx` before the M104 command resolves this.
Like this:
```
...
G90 ; use absolute coordinates
M83 ; extruder relative mode
Tx
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
...
```

TheMrCaveman, 2020-10-19:
strange, I do not have this problem. I can start single filament print from octoprint.
I have MK3S MMU2S 3.9.1 and 1.0.6.

; generated by PrusaSlicer 2.3.0-alpha1+win64 on 2020-10-19 at 07:16:04 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M73 P0 R34
M73 Q0 S34
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E4.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
;TYPE:Custom
M862.3 P "MK3SMMU2S" ; printer model check
M862.1 P0.4 ; nozzle diameter check
M115 U3.9.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
Tx
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

EddyMI3d, 2020-10-19:
Maybe I have a plugin that mixes it up, or it's because I have a MK2.5S.
Just wanted to share it if someone else has this issue...

aphasic1, 2020-10-19:
I am also having this problem, and I do not have a MMU2S, just a MK3S.  I sliced an object with 2.3.0 alpha and the print failed with M112. Looking through my octoprint logs, there are a LOT of communication errors between printer and octoprint.  It seems like they are out of sync in what line has been sent and Rec'd.  I tried the print twice and it failed at different points both times.  Once it just drew the skirt and outline of the object, another time it did half of the bottom infill before M112.  I don't think the solution (at least for my problem) is just inserting a simple Tx.  This may be a different issue than yours, but I am also failing with M112 in 2.3.0 alpha. 

```

| Send: N160 G1 X209.671 Y53.433 E0.03686*87
| Recv: Full RX Buffer
| Recv: ok
| Send: N161 G1 X209.671 Y165.091 E3.50097*109
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 155
| Recv: Resend: 156
2020-10-18 14:20:57,522 - octoprint.util.comm - INFO - Ignoring resend request for line 156, that still originates from lines we sent before we got the first resend request
2020-10-18 14:20:57,575 - octoprint.util.comm - INFO - Ignoring resend request for line 156, that still originates from lines we sent before we got the first resend request
2020-10-18 14:20:57,581 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 156, current line = 162
```

EddyMI3d, 2020-10-19:
@aphasic1 : Do you already updated the firmware to 3.9.1?

aphasic1, 2020-10-19:
@EddyMI3d  Yes, the firmware is 3.9.1.  Is this expected behavior for out of date firmware with 2.3.0-Alpha1? It's possible that I screwed something up when I updated the firmware.  My einsy hot end thermistor port died so I compiled the 3.9.1 firmware from source so I could change the hot-end thermistor port to the unused middle one.  The change just entailed switching a 0 to a 1 in the pins file, though, and it compiled fine and the printer reads 3.9.1 firmware and has been printing for 5 days straight since I flashed it (aside from the two failed attempts with a 2.3.0-alpha sliced object).

bubnikv, 2020-12-03:
Please discuss this topic with the firmware team. It is not PrusaSlicer's issue.
https://github.com/prusa3d/Prusa-Firmware/issues

closed 2020-12-03 by bubnikv