prusa3d/PrusaSlicer #4908: [2.3.0-Alpha1] Segfaults on startup

vintagepc, 2020-10-18:
### Version
2.3.0-alpha1

### Operating system type + version
OpenSuSE 42.3 x64

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
AppImage does not start; launching from a terminal shows it's crashing with a segmentation fault. 

I have tried both the standard and -GTK3 versions. I have also removed the .PrusaSlicer-alpha directory to ensure it was not a profile incompatibility with the Alpha0 build I had self-compiled. 

```
[2020-10-18 12:06:57.003125] [0x00007f63e56af580] [debug]   full path: 1776643031278729097
[2020-10-18 12:06:57.003163] [0x00007f63e56af580] [debug]   single instance: undefined. other params: /tmp/.mount_PrusaSWvvv52/usr/bin/bin/prusa-slicer;--loglevel;4
[2020-10-18 12:06:57.003179] [0x00007f63e56af580] [debug]   full lock path: /home/user/.PrusaSlicer-alpha/cache/1776643031278729097.lock
[2020-10-18 12:06:57.003211] [0x00007f63e56af580] [info]    instance check: Another instance not found or single-instance not set.
Segmentation fault (core dumped)
```

non-GTK3 reports this:
```
*** Error in `/tmp/.mount_PrusaSRc1Fvg/usr/bin/bin/prusa-slicer': malloc(): memory corruption: 0x0000000003537c40 ***
Segmentation fault (core dumped)
```

_Is this a new feature request?_

Bug report.

vintagepc, 2020-10-18:
This seems to be an issue with libpng. I've tried building locally and it throws a segmentation fault during compilation of libpng. 

Similarly, `strace prusa-slicer` reports the last thing it did was try to read a .png file:

```
openat(AT_FDCWD, "/usr/share/icons/oxygen/base/16x16/actions/gtk-cancel.png", O_RDONLY) = 12
fstat(12, {st_mode=S_IFREG|0644, st_size=848, ...}) = 0
futex(0x7f52c39a9428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7f52c39a9428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
read(12, "\211PNG\r\n\32\n\0\0\0\rIHDR\0\0\0\20\0\0\0\20\10\6\0\0\0\37\363\377"..., 65536) = 848
read(12, "", 65536)                     = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x20} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

```

vintagepc, 2020-10-18:
Issue found - The binary does not appear to be using the statically built libpng in deps/. 

Building locally and configuring with SLICER_STATIC as instructed in the .md exhibits the same segmentation fault. 

A stack trace in GDB revealed that it was still using the system libpng.so.16 in /usr/lib instead of statically linking the version built in deps/....

Updating the system libpng from 16.8 to 16.34 resolved the crash and the AppImage starts correctly now. 

There may be a similar issue with wxWidgets; the GTK3 version of the AppImage has a blank plater area (grey, no bed, cannot interact with it). I have to use the GTK2 version.

lukasmatena, 2020-10-18:
Thank you very much for the investigation. We'll try to fix it.

crami, 2020-10-21:
I see the same issue on debian/sid strace also shows the last thing is loading a png file

tamasmeszaros, 2020-10-23:
I've investigated the issue. PrusaSlicer links to libpng built statically among the provided dependencies. At the same time it links to libgtk2 or 3, depending on the wxWidgets configuration. That also links to libpng indirectly through libcairo. Static symbols are preferred during link time so the symbols from the static and shared libpng get mixed up, and all hell breaks loose. 

Ideally, it would be enough to just link to the libpng provided by Linux userspace, this was tried, did not work well for all cases, due to version incompatibilities. So the solution may be to build the static libpng with a prefix (through PNG_PREFIX var in its cmake) and link that to wxWidgets and ultimately to libslic3r. This way, cairo still uses its own instance of libpng, but it doesn't clash with the static instance.

2.3.0-alfa2 will contain the new solution, please check if it works for you after the release.

bubnikv, 2020-10-24:
It should be fixed in PrusaSlicer 2.3.0-alpha2. Closing.

closed 2020-10-24 by bubnikv