prusa3d/PrusaSlicer #4909: MMU2s Layer swap issue

MadMonkey0101, 2020-10-18:
### Version
_2.3.0-alpha1_

When slicing a model using mmu but its a single STL, i want to add extruder changes at layers, it allows me to select it, but when sliced there is no purge block shown, sliced same model same way in 2.2 and works fine

rtyr, 2020-10-18:
You have most likely wrong printer profile selected. Wipe tower was never used with "Single mode" printer profile.

[image]

NickOne81, 2020-10-18:
It is really flawed. Despite the correct printer selection with MMU, there is no filament change. It is still displayed in the preview, but after slicing it only prints with the color selected first.

rtyr, 2020-10-18:
I see, I was able to reproduce it. Thank you for reporting it.

wnc2, 2020-10-19:
same problem for me 
[image]

bubnikv, 2020-10-24:
It will be fixed with PrusaSicer 2.3.0-alpha2. Closing.

closed 2020-10-24 by bubnikv