prusa3d/PrusaSlicer #4910: Cannot mirror support enforcers (even though the option exists)

jt-github, 2020-10-18:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 v2004 build 19041.572

### Behavior
While a printable object (along with all of its support enforcers) can be mirrored (right-click the object, click Mirror, select an axis), support enforcer objects cannot be mirrored individually or in groups, even though the Mirror option still appears when right-clicking an enforcer.

This is not a new feature request since the menu option exists; this is a bug since nothing actually happens when attempting to mirror a support enforcer object (or group of them).

[Young__Red_Dragon.zip](https://github.com/prusa3d/PrusaSlicer/files/5398383/Young__Red_Dragon.zip)

jt-github, 2020-10-18:
Mirroring support enforcers would be extremely useful when working with symmetric (or mostly symmetric) parts, such as the attached dragon.  Normal auto-generated support "everywhere" creates a massive amount of extraneous supports on the model, hence the need for support enforcer objects.

lukasmatena, 2020-10-19:
>  this is a bug since nothing actually happens when attempting to mirror a support enforcer object 

As far as I can tell, that is not a bug. The enforcer is mirrored in its coordinate system, exactly as mirroring works for object parts. You can hardly tell for symmetric objects that you are using of course. Load a custom enforcer (asymetric stl) and see. I agree that whether this is the expected action with group selection might be questionable.

I suggest you try the new 2.3.0-alpha1 for now, the enforcers can be painted on the mesh which at least makes the job faster.

jt-github, 2020-10-19:
Oh, wow, painting on the mesh would be a fantastic improvement!  I'll have to try that out.

Just to clarify, you're saying that enforcers are automatically (but invisibly) mirrored... but across what axis?  If this is the case, then _that_ would certainly be a bug because it would cause issues on asymmetric parts.  Either way, this bug isn't terribly important for my purposes if the on-mesh painting method works well.

neophyl, 2020-10-19:
No, if you add a cube shaped enforcer and mirror it then it is mirrored on its local coordinate space.  However when you mirror a cube you get - ta da a cube.  Its relative to its own center location and not to the build plate position.  KInd of the same as rotating a modifier.   One you cant tell from the original one.  
I think Lukas is saying to load in a custom shaped model as a modifier, say a triangle, then mirror that and see what happens.

jt-github, 2020-10-19:
Doh!  Thanks for the explanation, @neophyl.  Now I understand what @lukasmatena was trying to tell me.  Mirroring works, but only on those simple objects... with essentially no net effect.

So, modifying my request/complaint, it's really about mirroring _groups_ of support enforcer objects.  This group-based mirroring could help anyone who wants to use "classic" support enforcer objects on a symmetric object so they can reduce the amount of setup time required.  That said, I'm hoping the whole painting-on-the-model thing works better for me anyhow.

bubnikv, 2020-12-07:
Do you need that feature? Or are you happy with the paint-on supports?

It is not such a problem to add a function to do the mirroring you wish, however it is difficult to present it to the user.

jt-github, 2020-12-07:
The paint-on will probably work instead, but I can still see situations where the mirroring would be useful (and maybe even more efficient and/or intuitive than the paint-on approach).  As for presenting it to the user, I would suggest a checkbox that appears whenever more than one object is selected with a label like “Mirror as group” that, when checked, flips the entire group over rather than flipping each object individually as it does today.  Obviously, when not checked, it should act just like it does now.

 

- John

closed 2026-09-29 by prusaslicer-bot[bot]