prusa3d/PrusaSlicer #4911: [2.3.0-Alpha1] Linux GTK3 cuttoff widgets

timothyhollabaugh, 2020-10-19:
### Version
2.3.0-alpha1+linux-x64-GTK3

### Operating system type + version
Archlinux, running XFCE

### 3D printer brand / version + firmware version (if known)
Reprappro Tricolour Mendel running Marlin

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

vintagepc, 2020-10-19:
I did observe this as well, though in my case the GTK3 version was unusable due to a completely blank "3d" area (plater, preview, placement, etc)

VorpalBlade, 2020-10-25:
I see the same using Cinnamon on Ubuntu 18.04. In addition some of the titles at the top were cut off in the configuration wizard:

[image]
[image]
[image]

pmjdebruijn, 2020-10-25:
I already issued a pull request with a potential fix: https://github.com/prusa3d/PrusaSlicer/pull/4490

With regard to the ConfigWizard this might be related: #4861

VorpalBlade, 2020-10-25:
I took a quick look at building PrusaSlicer myself (my day job is as a software developer on Linux) so I could test your PR, but unfortunately it was not as straight-forward as I had hoped, and would involve installing and building many dependencies. I do not currently have the time to do that. And it doesn't seem like PrusaSlicer is set up to build the AppImage with a CI either, which would make it a lot easier to just copy those steps.

VorpalBlade, 2020-11-23:
I was planning to test this now on Alpha4, but it seems there is no longer any GTK3 AppImage available to download. Thus I checked on alpha3 instead, and the issue is still there.

bubnikv, 2020-11-23:
Oh, I forgot. Let me build and post one at the release page.

VorpalBlade, 2020-11-29:
The issues with formatting in the GTK3 GUI remain in 2.3.0-beta1, though greatly reduced.

As can be seen, the plus buttons are somewhat cut off:
[image]

There also seem to be some random graphical corruption in the "Ironing Type" dropdown:
[image]

Settings loaded from a project file are cut off and formatted a bit silly (why is the file name in it twice?):
[image]

YuSanka, 2020-11-30:
@VorpalBlade, please retest with this build https://drive.google.com/file/d/10aknwdpHRei7oskIDspv9Y3sH0mCxC3q/view?usp=sharing

VorpalBlade, 2020-12-01:
> @VorpalBlade, please retest with this build https://drive.google.com/file/d/10aknwdpHRei7oskIDspv9Y3sH0mCxC3q/view?usp=sharing

This fixes the issues. (Apart from the too long profile name not fitting, which is not GTK3 specific. I don't know how that specific profile ended up with such a long silly name, and I cannot reproduce it in new files I create.)

VorpalBlade, 2020-12-01:
I found another issue though in your build: 
[image]

Notice black text on black background for the simple/advance buttons.

YuSanka, 2020-12-03:
Thanks for all your reports. Those bugs are fixed and will be a part of upcoming release.

closed 2020-12-03 by YuSanka