prusa3d/PrusaSlicer #4913: Details lost when slicing

ftrofin, 2020-10-19:
### Version
2.2.0+

### Operating system type + version
Mac 10.14.6
KDE Neon 20.0

### 3D printer brand / version + firmware version (if known)
Prusa Mini (stock)

### Behavior
* When slicing this object, detail is lost
* Open attached STL file in Prusa Slicer
* Look at the objects. Notice the concentric rings inside them
* Click Slice
* Look at the preview. No more circles. See difference:

[image]

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Roof_Fan_Covers.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5399738/Roof_Fan_Covers.3mf.zip)

jweaver100, 2020-10-19:
Not a bug.. Enable "detect thin walls" and you will get the circles.

jweaver100, 2020-10-19:
[image]

[image]

ftrofin, 2020-10-20:
Are there any downsides to having "Detect thin walls" turned on all the time? I presume so, otherwise this would be on by default I expect...

JohnOCFII, 2020-10-20:
> Are there any downsides to having "Detect thin walls" turned on all the time? I presume so, otherwise this would be on by default I expect...

There may be, but I do have it turned on in my profiles.  Only if something looks weird in preview do I turn it off.

closed 2026-09-29 by prusaslicer-bot[bot]