prusa3d/PrusaSlicer #4915: Won't Open - Windows 10

Schuetzenhaus, 2020-10-19:
### Version
2.3.0 alpha

### Operating system type + version
Windows 10, 64 bit

### Behavior
Doesn't open. I double click, run as admin, no help. It looks like something starts loading (little circle around the mouse cursor starts) but then stops. Also tried launching the console, it opens for a few seconds but then auto closes as well. The regular 2.2 version can open fine, it's just the 2.3 alpha.

lukasmatena, 2020-10-19:
Could you please run the application from the console with `--loglevel=5` command line argument? It should log something into the console so we can tell where it terminated.

Schuetzenhaus, 2020-10-19:
Thanks for the quick reply, how would I got about doing that?

lukasmatena, 2020-10-19:
Open command line (Win+R, type `cmd`, enter), `cd` into the directory with the binary (e.g. `cd C:\path\where\PrusaSlicer\lives`) and run it as `prusa-slicer-console.exe --loglevel=5`.

Schuetzenhaus, 2020-10-19:
Nothing happened - copy/pasted from cmd
C:\Program Files\PrusaSlicer-2.3.0-alpha1+win64-202010171250>prusa-slicer-console.exe --loglevel=5

C:\Program Files\PrusaSlicer-2.3.0-alpha1+win64-202010171250>

lukasmatena, 2020-10-19:
Thanks. It looks like it terminates fairly quickly and maybe outputs just an extra line break. Could you try to move it so the path does not contain the space in *Program Files* (e.g. just `C:\PrusaSlicer-2.3.0-alpha1+win64-202010171250`) and try again?

Schuetzenhaus, 2020-10-19:
Nada, same thing. https://puu.sh/GEKbk/a0185f123b.png

enricoturri1966, 2020-10-21:
@Schuetzenhaus
Have you already tried to delete the content of:
C:\Users\<...>\AppData\Roaming\PrusaSlicer-alpha
?

Schuetzenhaus, 2020-10-21:
> @Schuetzenhaus
> Have you already tried to delete the content of:
> C:\Users\ ... \AppData\Roaming\PrusaSlicer-alpha
> ?

That worked! Thank you so much

closed 2020-10-21 by Schuetzenhaus

lukasmatena, 2020-10-21:
@Schuetzenhaus I'm glad that you sorted the problem out, but could you please recover the directory from trash, zip it and send it here? Unless it contains some data that you don't want to share (Octoprint IPs, passwords, things like that).
Deleting people's config is not exactly the kind of advise we would like to tell our customers. And this is not the first issue like that.
Did you try any previous alphas (2.0, 2.1, 2.2), or was this your first one?

Schuetzenhaus, 2020-10-21:
[Alpha Files.zip](https://github.com/prusa3d/PrusaSlicer/files/5417729/Alpha.Files.zip)
Sure, here you go

This is my first alpha. I was running it quite a while ago when it was first compiled and worked great, then one day just stopped opening.

bubnikv, 2020-10-22:
We used to open this dialog if the PrusaSlicer.ini got corrupted. This error reporting got broken in PrusaSlicer 2.3.0-alpha1.

Hopefully this error message will be back with 027ab4fb022332a1cd9d7afec22297fc75cf69d9.

Do you have an idea how that PrusaSlicer.ini could have been broken? Did you experience any computer crashes recently?

bubnikv, 2020-10-22:
[image]

bubnikv, 2020-10-22:
[image]

Schuetzenhaus, 2020-10-23:
Mmm no, I don't think so. I don't remember any crashes or anything like that.

I think once I accidentally shut off my computer by hitting the power button my power brick so it shut the computer off unexpectedly, but I don't remember if I had prusaslicer open when that happened. Otherwise everything was normal.

bubnikv, 2021-07-02:
Transactional saving of PrusaSlicer.ini implemented in PrusaSlicer 2.3.2-beta.