prusa3d/PrusaSlicer #4916: [2.3.0-Alpha1 ] no infill  when 0 top layer selected

aschor, 2020-10-19:
### Version
2.3.0-Alpha1 

### Operating system type + version
windows 10 64b

### 3D printer brand / version + firmware version (if known)
corexy

### Behavior
if you select 0 top layer, then no infill is generated whatever the infill % selected

with 0 top layer : 
[image]

with 1 top layer :
[image]

lukasmatena, 2020-10-19:
I cannot reproduce that. Could you please provide a 3MF in the faulty state?

aschor, 2020-10-19:
here you are : 

[CoreXY_Calibration_Octagon_R1.zip](https://github.com/prusa3d/PrusaSlicer/files/5401006/CoreXY_Calibration_Octagon_R1.zip)

aschor, 2020-10-19:
@lukasmatena hummm nevermind, it's because of a setting in "infill" called "only infill where needed" that was checked. sorry for that

closed 2020-10-19 by aschor