prusa3d/PrusaSlicer #4917: Save G-code issue

Pajaentr, 2020-10-19:
### Version
2.2.0+win64

### Operating system type + version
Win10 Home, 1909

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega

### Behavior
If I press button to remove the SD card from PC before the .gcode file is saved, PrusaSlicer shows an error instead of wait for the file to be saved and then removing the SD card.

error: "Kopírování dočasného G-coru do výstupního G-codu se nezdařilo. Není SD karta chráněna proti zápisu?"

bubnikv, 2020-10-19:
> If I press button to remove the SD card from PC before the .gcode file is
saved

Which button? In Windows Explorer? Or slicer?
Does it happen with PrusaSlicer 2.3.0-alpha1?

Pajaentr, 2020-10-19:
Button in Slicer.
Here is image with steps in PrusaSlicer 2.2.0
[image]

and PrusaSlicer 2.3.0
[image]

kocikdav, 2020-10-19:
Thank you for reporting, we will investigate this problem. Is this happening on any model or a particular one? I see "neplatna data", thats why Im asking.

bubnikv, 2020-10-19:
Would you please try with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1
for us? Thanks.

Pajaentr, 2020-10-19:
I already tried it with this version. There is a picture with steps in one of my previous posts (one image/manual for version 2.2.0 and one for 2.3.0).

kocikdav, 2020-10-25:
@Pajaentr Unfortunately we were not able to reproduce this problem. But there are some changes in alpha2 that might suppress showing eject notification too early. Would you please try it?

Pajaentr, 2020-10-25:
Error reproduced with 2.3.0 alpha2
[https://youtu.be/TyVz1LchzwY](url)

kocikdav, 2020-10-26:
Aha, now I understand. Thanks for video. Should be repaired in next alpha.

closed 2020-11-07 by kocikdav