prusa3d/PrusaSlicer #4918: No way to remove filament types (functionality changed from 2.2.0)

jweaver100, 2020-10-19:
Version
2.3.0-alpha1+win64

Operating system type + version
Windows 10

3D printer brand / version + firmware version (if known)
Prusa Mk3 and Mini

Behavior

In 2.2.0 the Filament pull down looks like this.

[image]

AT the bottom there is a ------Add/Remove Filaments ------- option that you use to change which Filaments you have.

But in 2.3.0 you no longer have this option.

[image]

You can click at the top of the list where it says -------System Presets -------- and this launches the Config Wizard.. This is new functionality and I am not sure its intentional.

Here you have an option for 'Filaments', so perhaps this is how you do it... But the list of filaments is unticked and there is no realationship between this screen and what is in your "Filament" pull down.

Worse still, you CAN ticket filament types here.. And they get added.. But I can't find a way to remove them.... As you can't untick them (As they are already unticked when the screen opens).

Through playing around, I now have EVERY filament type enabled and no-way to remove them.. So it looks like I am going to have to blow away my config and start again.

jweaver100, 2020-10-19:
I thought I had found a way.. The PrusaSlicer.INI file has a list of filaments.. All set to 1.. So I changed them to 0 other than the ones I need (generic types)

Generic ABS @MINI = 0
Generic FLEX = 1
Generic FLEX @MINI = 0
Generic HIPS = 0
Generic HIPS @MINI = 0
Generic PETG = 1
Generic PETG @MINI = 1
Generic PLA = 1
Hatchbox ABS = 0
Hatchbox ABS @MINI = 0
Hatchbox PLA = 0
Plasty Mladec ABS = 0

Buit it made no difference. All of the Filaments are still listed.

YuSanka, 2020-10-19:
Thanks for your report. This bug is fixed and will be a part of upcoming release

commit 0f8f412e (2020-10-19): Fixed #4918
Added missed "Add/Remove filaments" item for the filament preset combobox
closed 2020-10-19 by YuSanka