prusa3d/PrusaSlicer #4919: [2.3.0-Alpha1] Graphical glitches when scrolling

vintagepc, 2020-10-19:
### Version
2.3.0-alpha1 (GTK2 version)

### Operating system type + version
OpenSuSE x64 42.3

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior

When scrolling in many areas the scrolled region does not always redraw correctly. (especially when the content moves up) I have had this happen in combo boxes, text areas, and the right hand panel (when set up in MM so that there are many extruder/filament combo boxes).  It's particularly annoying when it happens in text areas like the numerous start/stop/custom g-code boxes.

[image]

[image]

[image]

_Is this a new feature request?_

Bug/regression

bubnikv, 2020-12-07:
Do you still have this issue? If no, would you please close it?

vintagepc, 2020-12-07:
I have still seen it with the Beta1. Have not had a chance to test wiht Beta2 yet.

bubnikv, 2020-12-07:
Is it our binary build? Or did you compile yourself?

vintagepc, 2020-12-07:
It is the official binary from the release pages

vintagepc, 2020-12-10:
Just tested with the beta2. It does not seem to happen anywhere near as frequently, though I have still had it occur. 

[image]

rtyr, 2021-01-28:
Do you still have this issue?

vintagepc, 2021-01-28:
I have not seen it recently after starting to use the final 2.3 release.

rtyr, 2021-01-28:
So do you think we can close this?

vintagepc, 2021-01-28:
Let me check the GTK2 build after work tonight and see if I can deliberately make it happen again; if not then I agree there isn't any reason to keep it open. 👍

vintagepc, 2021-01-29:
Does indeed look like it's been resolved.

closed 2021-01-29 by vintagepc