prusa3d/PrusaSlicer #4920: Feature request: allow shapes to subtract as well as add.

towlerg, 2020-10-19:
### Version
all but currently 2.3.0 Alpha

### Operating system type + version
Win7 sp1 x64

### 3D printer brand / version + firmware version (if known)
Stock Sovol SV01, marlin, 2.0.6

_Is this a new feature request?_
Yes

Rational : I often find I only have an .STL file and it easier to make small mods (wider, stronger etc.) in Prusaslicer than any CAD I know. At the moment the only way to remove material is with cut. It would be neat if you could use shapes to remove as well as add.
Also, a triangle shape would be super.

lukasmatena, 2020-10-19:
You can remove material, although it is a bit hacky. Load a modifier mesh and set it to zero perimeters, zero infill, zero top and bottom layers and you'll see.
We may implement a more user friendly solution in future.

towlerg, 2020-10-20:
Actually not hat hacky at all. For my part I'm perfectly happy with your solution althought paint on remove would look really cool. 
BTW a cylinder is a great way to quickly add a hole

Any chance of a triangle shape?

As I'm happy with solution should I close this?

towlerg, 2020-10-20:
Don't seem to be able to edit my previous.  For hat read that.

lukasmatena, 2020-10-21:
> Any chance of a triangle shape?

I'm not sure. You can load any STL and use it this way, as I have shown e.g. in https://github.com/prusa3d/PrusaSlicer/issues/4349#issuecomment-639308791. I personally don't think that we need a generic triangle.

> As I'm happy with solution should I close this?

I changed the title and would leave it open, if anyone has anything to add to this.

mankyd, 2021-03-28:
I want a different boolean operator: print only what's contained within the modifier. This would make rapid prototyping that much more rapid. You could easily print just a small section of a larger object to see if any small adjustments are needed before committing to a larger print.

neophyl, 2021-03-28:
You can do that with the above modifiers already.

Set your perimeters to zero, top and bottom to zero and infill to zero.  You will then get an error when slicing saying an object has no extrusions on the first layer.  Then place a modifier over the part you want to print and set that to have whatever non zero perimeters, top, bottom etc you want.  When sliced it will just slice the area inside the modifier.

bubnikv, 2021-06-03:
Negative volumes aka subtractors are implemented in PrusaSlicer 2.4.0-alpha1.

closed 2021-06-03 by bubnikv