prusa3d/PrusaSlicer #4924: Feature request - Force inner perimeters to continue to outer perimeters

iso250, 2020-10-19:
### Version

2.3.0 Alpha 1

### Operating system type + version

MacOS Catalina

### 3D printer brand / version + firmware version (if known)

Mk3S

### Behavior

New feature request - Force continuous perimeter

Sometimes when printing perimeters the slicer will print an inner perimeter, retract, move the extruder, then unretract a new line for the outer perimeter instead of just continuing the inner perimeter on to the outer perimeter.

When printing drafts at high speeds the unretraction for the outer perimeter can late retract, leaving unsightly gaps in the outer perimiter.  If there was a toggle that would force the inner perimeter to continue to the outer perimeter then the late extrusion gap would be located inside the model where you couldn't see it.

Whilst drafts and prototypes need not be perfect for some reason large late extrusion gaps seem to affect clients' opinions.

When printing prototypes of drafts for clients to consider I have to sometimes print slower than I'd like in order to eliminate this gap.  That defeats the purpose of high speed drafts and prototypes.  By forcing the inner perimeter to continue to the outer perimeter you are hiding these gas, plus you are reducing the number of retractions and unretractions, thus speeding up the print.

neophyl, 2020-10-20:
Sounds like Super Slicers looping perimeter option.

bubnikv, 2020-10-20:
> Sounds like Super Slicers looping perimeter option.

The looping perimeter was first prototyped by Kisslicer.  Does it work
better than the regular for the real world prints?

> Sometimes when printing perimeters the slicer will print an inner
perimeter, retract, move the extruder, then unretract

Deretract / retract shall not happen for short travel moves.

neophyl, 2020-10-20:
I've not installed Kisslicer so I am not familiar with it.  I just thought the description by the original poster sounded like the looping perimeter function.

I don't really use it on my normal prints although I did have a play with it when it was added. For me it produced a slightly more noticeable seam artefact, it bulged out slightly more.  Then again it was on a cr10 with a 0.6mm nozzle that doesn't do great seams anyway.  It was faster at printing so I guess if you are doing drafts it wouldn't make much difference.  I also didn't try and tune it any.  Actually that entire printer needs tuning anyway as its my project and draft printer.  Precision prints are what my Mk3 is for :)

iso250, 2020-10-20:
> Deretract / retract shall not happen for short travel moves.

It's ending the inner perimeter, moving more than 3 cms (more than the "Minimum Travel After Retraction setting) and then starting on the outer perimeter.

>  It was faster at printing so I guess if you are doing drafts it wouldn't make much difference.

My seams are razor sharp, but there is a touch of ringing in the print.  Funny thing is that my clients aren't fussed with seeing those on a draft but they are fussed seeing the gaps that are left when speed printing leaves a gap at the beginning of the extrusion!

iso250, 2020-10-20:
SuperSlicer doesn't have the feature I want, but it can sometimes be replicated by using the Travel Move reduced option.  That will sometimes force the inner perimeter to continue to the outer perimeter, but not always.

bubnikv, 2020-10-20:
> It's ending the inner perimeter, moving more than 3 cms (more than the
"Minimum Travel After Retraction setting) and then starting on the outer
perimeter.

That should not happen for an aligned seam.

iso250, 2020-10-20:
> It's ending the inner perimeter, moving more than 3 cms (more than the "Minimum Travel After Retraction setting) and then starting on the outer perimeter. 

>That should not happen for an aligned seam.

That's what I thought, too.  The thing is that I have it set to aligned seam and it happens on PrusaSlicer 2.2.0 AND the Alpha build.  It doesn't happen all the time, but enough to be annoying.

iso250, 2020-10-20:
It's hard to see the retractions and unretractions with the new colour scheme, but here you can see that the inner perimeter (yellow perimeter) starts on the left and retracts on the left.  The extruder then moves to the middle, unretracts, traces the outer perimeter and then retracts near the middle.

By rights, the line should start on the left, trace out the inner perimeter and then start the outer perimeter on the left as an extension of the inner perimeter, with the only retraction happening on the outer perimeter on the left.

[image]

rtyr, 2021-02-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3328.

iso250, 2021-02-04:
> Similar to #3328.

Almost everything I print has walls with no fill. That would require the extrusion line to pass through the wall to complete a layer and that’s not something I want. It would weaken the walls and there’s the likelihood that thicker nozzles would create bulges when this passthrough happens. 

I’ve tried looping perimeter on super slicer and with larger nozzles the results are never as neat as properly tuned retraction.

closed 2026-09-29 by prusaslicer-bot[bot]