prusa3d/PrusaSlicer #4926: Feature Request: add custom models to the built-in "lambda" models

kburkhart84, 2020-10-19:
I like the lambda model features, though the current 4 included models are limited.  They do cover lots of use cases, and I like how they can be used as actual sliced parts or as modifier sections.  However, I'd like to be able to add my own parts to them.  Yes, I know I can load parts from external stls just like models, which can be used as modifiers as well.  However, I'd like to have maybe a directory which would contain my own models, and then be able to easily add them from the same shape adding menu.

I'd use this primarily for utility parts that get used consistently.  For example, every flat keychain lithophane I make I add a square with a hole to one of the corners.  Right now I simply add it in as an additional part, but I'd like to be able to just see it on that "Add-part" menu so I'm not having to navigate directories.  i'm sure this would come in handy for many people as there are many use cases.

bubnikv, 2020-10-20:
I believe you posted your Slic3r issue to PrusaSlicer gihtub tracker by mistake.

bubnikv, 2020-10-20:
Or maybe not?

foreachthing, 2020-10-20:
Why not just add the path to "Quick access" (Windows) or to Bookmarks (Linux)?
I'm sure one can do that on macOS as well.

kburkhart84, 2020-10-20:
Oh yes, it is meant for Prusaslicer, though it may not be using the "lambda" terminology for the parts anymore.

So yes, doing a quick access should work, however I still think it would be even nicer and quicker if it had integration directly in the menus. It could be a subfolder next to the cube, cykinder, etc... and could even have subfolders within that based on the file structure in that folder.

bubnikv, 2021-06-03:
Implementing this will make these people happy https://github.com/prusa3d/PrusaSlicer/issues/4349 as well I suppose

rtyr, 2021-08-12:
Implemented. Shape gallery will be available in the upcoming PrusaSlicer 2.4.0-alpha1. Closing.

[image]

closed 2021-08-12 by rtyr