prusa3d/PrusaSlicer #4932: Feature Request: Support Enforcers Should Turn On Supports When Added

wjsteele, 2020-10-21:
Feature Request:

When support enforcers are added to a model, the setting for supports should be switched to at least "For Support Enforcers Only."  Right now, a lot of users miss that setting and don't discover it until after they've already started a print.  If you're adding support enforcers, naturally, you'll want supports enabled.

bubnikv, 2020-12-07:
We are considering to add a notification with a hyperlink to switch from "no supports" to "for support enforcers only" as we now have the infrastructure for the notifications.

lukasmatena, 2021-03-04:
Implemented with abd5a9a46e16bab66988bf338516b1e0d1f5b4e8 for the next release:

[image]

Closing.

closed 2021-03-04 by lukasmatena