prusa3d/PrusaSlicer #4933: Prusa "MK3S" displays as "MK3" (no S) using Configuration over USB

Jaxx005, 2020-10-21:
### Slicer Version: 2.2.0+win64
### Win10 x64 [Fully up-to-date.]
### Prusa i3 MK3S

### Behavior
Connect a MK3S via USB to Win10 PC
Prusa Slicer:  select Configuration \ Flash Printer Firmware \ [rescan]
Serial port: displays "Original Prusa i3 MK3" 
The suffix "S" is not shown

NOTED:  
This may be a printer-derived title (it is definitely a 'S' version, and the the printers Info screen confirms this) and not a Slicer issue at all

With hindsight, there seems to be no functional issue here (firmware update successsful, in this situation at least), but users may query the 'mis-title' and spend time checking it is correct.  

The title should state "...MK3S..." or "MK3/MK3S..." to indicate it could be either model.

[image]

Regards
Lee

rtyr, 2020-10-21:
That's normal. PrusaSlicer just reports the device name assigned to selected port. MK3 and MK3S are both using the same device name without "S". Closing.

closed 2020-10-21 by rtyr

Kachidoki2807, 2020-10-22:
> That's normal.

I don't want to re-open this issue, but only expose my though. I fully understand the explanation from a technical point of view, I am not sure this behavior can even be changed.

But from the lambda user point of view, that is **not** normal.
Would you agree to buy an "Ultimate Super Cool Device" an found it named "Super Cool Device"? No, you just bought an Ultimate, it must be displayed Ultimate.

Moreover, it could lead to some misunderstanding and to a bad user experience during this critical step which is flashing a new firmware. Indeed, a newbie will absolutely take care of which firmware he will download, but once facing this enumeration, he will be in trouble, thinking he did something wrong.

Jaxx005, 2020-10-22:
> Moreover, it could lead to some misunderstanding and to a bad user experience

That is exactly what happened in my case: I stopped and double-checked there wasn't another FW version that should have been downloaded. So yes, some confusion. But it did work out OK (but a total Newbie might not have been so lucky).

Unlikely scenario, but... if the S firmware differs, then I suppose someone owning both version could mistakingly connect an S version and incorrectly install the non-S FW.

I reported for Newbies - that and it seemed like an easy fix.  :(

rtyr:  So this is a 'printer' issue?  Can this be forwarded perhaps?  Leave it with ya ;)  Thanks.