prusa3d/PrusaSlicer #4934: Wait for first layer temp between sequential objects

combs, 2020-10-21:
### Version
2.3.0-alpha1

### Operating system type + version
lunix!
Ubuntu 20.04 on amd64

### 3D printer brand / version + firmware version (if known)

Real Original Prusa mk3 -> mk3s

### Behavior

When printing multiple sequential objects with a higher first layer temperature, the printer does not wait for the first layer temperature to be reached before beginning to print subsequent objects.

This leads to adhesion issues for object 2+ in cases where there are very different first layer vs. subsequent layer temperatures set.

Example--first layer set to 210 and subsequent layers set to 190:

- Printer waits to reach 210
- Printer prints first layer
- Temperature naturally falls to 190 over course of the print
- Printer begins to heat to 210
- Printer moves and begins printing immediately (at say 192 degrees) before reaching 210

As a workaround I added the following gcode to "Between objects G-code" field and am seeing much improved adhesion for 2nd+ objects.

```
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

_Is this a new feature request?_
I think it's probably an oversight/bug 

#### Project File (.3MF) where problem occurs
[example.zip](https://github.com/prusa3d/PrusaSlicer/files/5417867/example.zip)

combs, 2020-10-21:
here's a live example video, the temperatures are 215 (first layer) and 195 (subsequent layers). You can see it begins to print at 196 degrees despite being set to 215

[wait-first-layer-temp-subsequent-objects.zip](https://github.com/prusa3d/PrusaSlicer/files/5417928/wait-first-layer-temp-subsequent-objects.zip)

neophyl, 2020-10-22:
One thing I noticed in the 2.3.0 alpha is there is a custom gcode section now for between objects for consecutive printing.  If you were to use that and add in the M109 S[first_layer_temperature] ; wait for extruder final temp gcode it should set it back to the first layer temp between objects.

combs, 2020-10-22:
@neophyl Yeah! I mentioned in the issue that this worked well as a workaround. 

Perhaps this G-code was hard-coded before 2.3.0-alpha1?

bubnikv, 2020-12-07:
You should be able to script your temp changes in 
between_objects_gcode
custom G-code. Just place your temp commands there and the defaults will not be emitted.

combs, 2020-12-07:
> You should be able to script your temp changes in
> between_objects_gcode
> custom G-code. Just place your temp commands there and the defaults will not be emitted.

Yes, I mentioned in the issue text that you can do that as a workaround. It should be the default behavior, in my opinion.

superbole, 2021-01-08:
Does anyone know where we can find more information on the "between_objects_gcode"? For example the variables that can be used?

If we could, for example set a different temperature per object we would be able to do a temperature test per object (or for different instances of an object) instead of having to print a temperature tower.

foreachthing, 2021-02-23:
> Does anyone know where we can find more information on the "between_objects_gcode"? For example the variables that can be used?

I don't know how up to date this page is but it has helped me a lot: http://projects.ttlexceeded.com/3dprinting_prusaslicer_gcode.html

And as always: https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)

closed 2026-09-29 by prusaslicer-bot[bot]