prusa3d/PrusaSlicer #4935: Gtk-CRITICAL messages

andy9a9, 2020-10-21:
### Version
2.3.0-alpha1+linux-x64-202010171312

### Operating system type + version
Ubuntu 18.04

### Behavior
Automatically after application lot of GLib critical errors is printed.

```

(prusa-slicer:24146): Gtk-CRITICAL **: 20:04:36.709: IA__gtk_window_resize: assertion 'width > 0' failed

(prusa-slicer:24146): GLib-GObject-WARNING **: 20:04:36.751: invalid (NULL) pointer instance

(prusa-slicer:24146): GLib-GObject-CRITICAL **: 20:04:36.752: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:24146): GLib-GObject-WARNING **: 20:04:36.752: invalid (NULL) pointer instance

(prusa-slicer:24146): GLib-GObject-CRITICAL **: 20:04:36.752: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:24146): GLib-GObject-WARNING **: 20:04:36.762: invalid (NULL) pointer instance

(prusa-slicer:24146): GLib-GObject-CRITICAL **: 20:04:36.763: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

...

(prusa-slicer:24146): GLib-GObject-WARNING **: 20:04:37.153: invalid (NULL) pointer instance

(prusa-slicer:24146): GLib-GObject-CRITICAL **: 20:04:37.153: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

lukasmatena, 2020-10-21:
Thanks, we know about them. However, as far as we know, they are harmless and to be honest, we have more important things to do than getting rid of them (if that is even possible). Also: https://github.com/prusa3d/PrusaSlicer/issues/4691#issuecomment-684831967

bubnikv, 2020-10-22:
In general, this kind of warnings is pretty common for other GTK applications as well. I am not saying that this is a correct situation, but maybe GTK is being a little bit too picky.

martinberlin, 2021-02-03:
I have the same as here but my PrusaSlicer does not open any more in 2.3.0 Any solution?
First I got: 
Gtk-Message: 12:32:47.899: Failed to load module "canberra-gtk-module"

So I installed it in Ubuntu 20:
The following NEW packages will be installed:
  libcanberra-gtk-module libcanberra-gtk0

And then after this got this error. Confused, didn't happened with older versions

from-nibly, 2021-03-16:
My prusa slicer wont start. I'm getting the following output when running on ubuntu linux

```
(prusa-slicer:865940): GLib-GObject-WARNING **: 09:21:27.916: invalid (NULL) pointer instance

(prusa-slicer:865940): GLib-GObject-CRITICAL **: 09:21:27.916: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:865940): GLib-GObject-WARNING **: 09:21:27.916: invalid (NULL) pointer instance

(prusa-slicer:865940): GLib-GObject-CRITICAL **: 09:21:27.916: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:865940): GLib-GObject-WARNING **: 09:21:27.916: invalid (NULL) pointer instance

(prusa-slicer:865940): GLib-GObject-CRITICAL **: 09:21:27.916: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:865940): GLib-GObject-WARNING **: 09:21:27.916: invalid (NULL) pointer instance

(prusa-slicer:865940): GLib-GObject-CRITICAL **: 09:21:27.916: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 617 error_code 2 request_code 152 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

it looks similar to the output above.

cgspeck, 2021-04-24:
I am having the same issues that @from-nibly is having attempting to use both versions of the latest AppImages on Ubuntu 20.04.

from-nibly, 2021-04-24:
I ended up rebooting and the error went away

closed 2024-11-21 by lukasmatena