prusa3d/PrusaSlicer #4936: External perimeter extrusion width absolute setting (mm) not honored.

greendog99, 2020-10-21:
### Version
2.3.0-alpha1

### Operating system type + version
MacOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Anycubic Mega-S

### Behavior
External perimeter extrusion width varies based on layer height, even if a fixed mm width is set in "Print Settings" -> "Advanced".

To reproduce:

* Set a fixed value (e.g. 0.45mm) for all "Extrusion width" settings in "Print Settings" -> "Advanced".
* Add a height range modifier to part of a model, or use Adaptive variable layer height.
* In the "Preview" pane, set the "View" to "Width". Layers with a different height have a different (not 0.45mm) external perimeter width and are displayed in a different color.
* In PrusaSlicer 2.2.0, these same steps result in all layers having the same (correct) external perimeter width regardless of layer height.

[image]

In the below preview, the layer height changes from 0.1mm to 0.35mm at the 10mm mark (via a height range modifier):

[image]

By using more extreme values (e.g. with a 1mm nozzle and 0.1 vs 0.8mm layers), it can be seen more clearly that the external perimeter width is changing in the preview, not just the color.

rtyr, 2020-10-21:
Reproduced. Thank you.

greendog99, 2020-11-22:
FYI this regression (worked fine in 2.2.0, appeared in 2.3.0-alpha1) is still present in 2.3.0-alpha4. Since this was tagged "g-code preview" I also wanted to note that I suspect this affects exterior perimeter printing as well as the preview. Thanks!

bubnikv, 2020-12-07:
Would you please verify whether this has been fixed in PrusaSlicer 2.3.0-beta2? Thank you.

greendog99, 2020-12-07:
Still broken in 2.3.0-beta2. In the screenshots below, I set my normal layer height to 0.35mm, and set a height range modifier for a layer height of 0.08 in the middle of the print. I set all extrusion widths to 0.45mm. It looks like PrusaSlicer is choosing ~0.40mm as the exterior perimeter width for the 0.35mm layer heights, and choosing ~0.48mm as the exterior perimeter width for the 0.08mm layer heights:

[image]

[image]

Thanks! Loving 2.3.0 in general and appreciate all the hard work being done to improve it!

bubnikv, 2020-12-12:
Fixed in PrusaSlicer 2.3.0-beta3

closed 2020-12-12 by bubnikv