prusa3d/PrusaSlicer #4937: Ironing flag ignored on infill rotate

5teveO, 2020-10-22:
2.3.0 alpha

### Operating system type + version
Windows 10 latest updates

### 3D printer brand / version + firmware version (if known) CR10MAX self tweaked tm3d dw7 firmware. Std printer otherwise

### Behavior
If ironing is switched on and you then change infill angle and slice, no ironing is included in the slice.

To get ironing back you have to turn ironing off, slice, then turn it on again and slice again. It then reappears.

The print time etc all indicate that ironing isn't present when it disappears.

I noticed this as I tend to run the longer length on prints along the X axis to save swinging the bed around as it is heavy. Depending on the number of layers this determines the angle of the ironing, therefore if it's not longitudinal along the X I rotate the infill from 45 to 135 degrees

5teveO, 2020-10-22:
Apologies this looks like another variation of issues already raised..  I should have looked back further..

bubnikv, 2020-10-22:
Likely a duplicate of #4510

closed 2020-10-22 by bubnikv