prusa3d/PrusaSlicer #4939: Monotonic o last layer before ironice

vvv850, 2020-10-22:
### Version
2.3.0 Alpha 1 Windows 64

### Operating system type + version
Windows 10 2004 x64 

### 3D printer brand / version + firmware version (if known)
MK3s Stock
3.9.1

### Behavior
Feature Request:
Add monotonic infill for the last layer before ironing.
Currently, even if you select monotonic, once you select ironing, it does not apply the monotonic feature on the last layer/

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2020-12-07:
Maybe just apply monotonic at every layer.

bubnikv, 2021-02-11:
e026ed87181b542bf5b2f90567d8ce06f05e1659

    Experimental: Switching all solid infills from Rectilinear to Monotonic
    for regions with Monotonic top infill type.

We will now evaluate internally whether switching all internal infill to Monotonic does not have some unintentional negative effect.

bubnikv, 2021-07-02:
Released with PrusaSlicer 2.3.2-beta.

closed 2021-07-02 by bubnikv

Pun-e, 2021-07-16:
I just wanna add a big thank you for implementing this, this is also helpful for people working with translucent and transparent materials as the slight imperfections are visible through the final top layer