prusa3d/PrusaSlicer #4940: Name of output files

JCD95-lab, 2020-10-22:
### Version
_Version of PrusaSlicer used goes here_2.2.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_windows 7 edition familiale service pack 1
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version Artillery Genius 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
 _Describe the problem_ i want to have the name of the property "fill_pattern" in the name of the output file

  I put the directive :  {input_filename_base}_{layer_height}mm_{filament_type[0]}_**_{fill_pattern}_**_{print_time}.gcode

the resulte is 
 Iron_Throne_Benchy_0.16mm_PLA_**(fill_pattern)**_54m.gcode

Best regards

neophyl, 2020-10-22:
try using **[fill_pattern]** with the square brackets.  That worked for me with 2.3.0 on windows.  In fact all my placeholders use the [] square brackets and not the {} ones.

JCD95-lab, 2020-10-23:
Than you, fonctionnary correct .

bubnikv, 2020-12-05:
solved.

closed 2020-12-05 by bubnikv

JCD95-lab, 2020-12-06:
Thank you, good information...😇😇😇