prusa3d/PrusaSlicer #4941: Feature Request: Lift Z when travelling in the same height as an incoming perimeter

Pizzahd88, 2020-10-22:
Version
PrusaSlicer 2.3 Alpha 1

Operating system type + version
Windows 10 x64

3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro / Marlin 2.0.6

Is this a new feature request?
Yes

Explanation:
I never use Lift Z, as it leaves more stringing in my prints. 
But the prints where I am forced to print them with Lift Z is something like a flat spring print (ex. the Modular spring here: https://www.thingiverse.com/thing:3338467/files). (see images below for example)

The reason for me needing Lift Z is that when the print head ends its print in the outer perimeter but still have something to print in the middle of the print before starting the next layer, it travels straight across all the circles. If we have just a tiny amount of plastic scratching the nozzle on one of the perimeters, it knocks them off the build plate.
In these cases a feature "Lift Z outside perimeter" would come in handy, as we would not lift z on normal printing inside the part, we would wait until we go across the perimeter and the nozzle is free from the print. While moving to the next perimeter we could raise Z (to avoid unnecessary stops).
When we reach the endpoint, we do the usual lower-Z logic while deretracting.

Sliced model with "Avoid Crossing Perimeters" DISABLED
[image]

And with "Avoid Crossing Perimeters" ENABLED:
[image]

Pizzahd88, 2020-10-22:
It could be a value instead of an on/off toggle.
Ex: "Lift Z Outside Perimeter" = 0.3 mm

brettvitaz, 2020-10-27:
Just a thought, but if your nozzle is knocking into your print, the printer may be over-extruding.

Pizzahd88, 2020-10-28:
It might be caused by a lot of things:
- A part of the spiral might have loosened from the build plate.
- Over extrusion
- Debrees loosened from the nozzle
- Uneven build surface
- x/y/z axis binding
- belt not tightened perfectly
- stepper motor issues
- a part of the machine might have become a bit loose after use causing a slight movement
- etc.

The reasons are many that can cause the issue, and the solution can help the print survive most of them, unless they are severe.

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

We wonder whether the reworked "avoid crossing perimeters" improves your issue?

jeremy-wendelken, 2021-04-22:
Does the  'Only retract when crossing perimeters' setting not achieve the desired effect?

[image]

Pizzahd88, 2021-04-22:
No that relates to the retraction. We want retraction at all times but lift z when crossing outer perimeters

jeremy-wendelken, 2021-04-22:
Why do you need to retract within infill as usually that part of the print will be hidden once complete?

Pizzahd88, 2021-04-23:
Because LiftZ lifts the nozzle which is different from retraction. 

Avoid nozzle to hit objects while outside the perimeter and traveling over other perimeters. Retracting does not achieve that. 

Lift Z inside perimeter is not wanted as it leaves dots all over the place and results in stringing sometimes. 

If we could use LiftZ only when going outside the perimeter we could avoid hitting objects on travels while not using LiftZ inside perimeter.

closed 2026-09-29 by prusaslicer-bot[bot]