prusa3d/PrusaSlicer #4943: mDns half working

Brandon2255p, 2020-10-23:
### Version
2.3.0-alpha1+win64

### Operating system type + version
Win 10 Pro

### 3D printer brand / version + firmware version (if known)
Irrelevant but using Klipper on custom Prusa i3

### Behavior
* Unable to use the mDns address for my printer. Testing the connection is successful. 
When trying to upload, it is not able to resolve the address. Is it using different code to resolve?

[image]

* give the printer an mDns address (octoprint does it by default)
  * enter the octoprint address in the Print Host Upload
  * Test connection. It should be successful
  * Try upload
* Expected the upload to be successful
* Actual Result:
[image]

_Is this a new feature request?_... maybe but seems more like a bug

cabbey, 2020-10-24:
Is it 100% repeatable for you? I get this host resolution error on upload about 1/3 to 1/2 of the time; never on test... but a retry is likely to go through after a failure, especially if I wait a couple minutes.

bubnikv, 2020-12-07:
Isn't it a duplicate of
https://github.com/prusa3d/PrusaSlicer/issues/1781
?

Brandon2255p, 2020-12-08:
@cabbey I have not seen any success so seems to be 100% repeatable. 
@bubnikv that issue does seem to be the same but it is less descriptive

kroltan, 2024-12-25:
I have a similar issue to @cabbey  on Linux, targeting PrusaLink. Name resolution fails 2/3 of the time. Sometimes it succeeds for the initial check when I click "send to printer", I get to type the filename, but fails subsequently when uploading the file, other times it fails outright.

`dig myprinter.lan` or any other software works reliably, so it's not my setup

closed 2026-09-29 by prusaslicer-bot[bot]