prusa3d/PrusaSlicer #4944: SD card ejection fails with integrated SD readers

leptun, 2020-10-23:
### Version
2.2.0+win64 - release

### Operating system type + version
Windows 10, build 2004.

### 3D printer brand / version + firmware version (if known)
MK3S development 3.9.1+. Unrelated to the issue.

### Behavior
Whenever I use the integrated microSD card slot on my laptop with PrusaSlicer, I get an error when I try to eject the card. Uploading works, the correct partition is detected (two of them, it's a rPI sd image). This eject issue only happens if the integrated reader is used. If I use any other SD reader, ejecting works. In my experience, this seems to be limited to ASUS laptops as a colleague of mine who has a different ASUS laptop (full sized SD reader) still has the same issue as me. If I try to eject the card from windows directly, it works regardless of reader, so it's most likely a PrusaSlicer issue. Had this issue since when SD card ejecting was implemented in PrusaSlicer a long time ago.
In device manager, the reader is named `Intel SD Host Controller`. Let me know if you want any other specific information about this as it looks like an edge case issue only affecting a small range of controllers.

_Is this a new feature request?_
Looks like it. At least on windows.

[image]

leptun, 2020-10-23:
Looks like this might be related to Intel SD controllers and it's not limited to Windows...
Related to https://github.com/prusa3d/PrusaSlicer/issues/3975

kocikdav, 2020-10-25:
Hello. Would you please try our latest released alpha version (2.3.0-alpha2). There were several changes in ejecting algorithm since 2.2.0 so there is a chance this problem is solved there.

leptun, 2020-11-02:
Sorry for the late response. For some reason I didn't get a notification of your comment.
Just tried the latest Alpha and it still fails.
[image]

kocikdav, 2020-11-19:
Hello. Unfortunately we were not able to find computer that would reproduce this issue. I have added some debug logging, maybe it will help us. Please try this build with --loglevel=5.
OSX: https://www.dropbox.com/s/hstwticcsjp1dqk/PrusaSlicer-2.3.0-alpha3%2B96-dk-eject-asan-ga4b9801b8-202011190849.dmg?dl=0
LINUX: https://www.dropbox.com/s/bkiqqlj3oub0xu7/PrusaSlicer-2.3.0-alpha3%2B96-dk-eject-linux-x64-asan-ga4b9801-202011190849.AppImage?dl=0
WINDOWS: https://www.dropbox.com/s/2t5dvuztl6f4yfn/PrusaSlicer-2.3.0-alpha3%2B96-dk-eject-win64-ga4b9801b8-202011190849.zip?dl=0

leptun, 2020-11-20:
@kocikdav I ran the windows build you sent from powershell with --loglevel=5.
```
[2020-11-20 09:07:43.216486] [0x00001e08] [info]    Ejecting started
[2020-11-20 09:07:43.229174] [0x00001e08] [debug]   FSCTL_LOCK_VOLUME 1 ; 0 ; 0
[2020-11-20 09:07:43.233184] [0x00001e08] [debug]   FSCTL_DISMOUNT_VOLUME 1 ; 0 ; 0
[2020-11-20 09:07:43.247903] [0x00001e08] [error]   Ejecting \\.\D: failed (IOCTL_STORAGE_EJECT_MEDIA)0 0
```

kocikdav, 2020-11-23:
Thank you very much for testing. Unfortunately Im just not able to find what might go wrong. I tried to leave FSCTL_LOCK_VOLUME and FSCTL_DISMOUNT_VOLUME from the code. Maybe they are blocking the ejecting. Can I ask you for one more test? 
https://www.dropbox.com/s/d7qmg6avfzbces6/PrusaSlicer-2.3.0-alpha4%2B9-dk-device-win64-g97ebc55f5-202011231948.zip?dl=0

leptun, 2020-11-24:
Still fails :/
```
[2020-11-24 08:10:21.443469] [0x00004d60] [info]    Ejecting started
[2020-11-24 08:10:21.446463] [0x00004d60] [error]   Ejecting \\.\D: failed (IOCTL_STORAGE_EJECT_MEDIA)0 0
```

kocikdav, 2020-12-02:
Hello, thanks. Well one more option is to try unlocking before eject with IOCTL_STORAGE_MEDIA_REMOVAL. Hopefully this one will work.
https://www.dropbox.com/s/sm86pyngotqbhpz/PrusaSlicer-2.3.0-beta1%2B48-dk-device-win64-gcfa5254ed-202012022110.zip?dl=0

bubnikv, 2020-12-03:
@leptun 
Does the eject function in Cura works on the same internal SD card reader? If no, then I am afraid it is a limitation of the SD card drivers and the only thing we can do is to document it (and maybe possibly detect the particular SD card reader).

leptun, 2020-12-03:
The last test still fails, but with a 3 second delay:
```
[2020-12-03 09:18:00.646339] [0x000054dc] [info]    Ejecting started
[2020-12-03 09:18:00.654542] [0x000054dc] [debug]   FSCTL_LOCK_VOLUME 1 ; 0 ; 0
[2020-12-03 09:18:00.661635] [0x000054dc] [debug]   FSCTL_DISMOUNT_VOLUME 1 ; 0 ; 0
[2020-12-03 09:18:00.669635] [0x000054dc] [debug]   IOCTL_STORAGE_MEDIA_REMOVAL 1 ; 0 ; 0
[2020-12-03 09:18:00.676640] [0x000054dc] [error]   Ejecting \\.\D: failed (IOCTL_STORAGE_EJECT_MEDIA)0 0
```
Cura 4.8.0:
[image]

I still find it very weird that windows has absolutely no problem ejecting the device, but programs do.

bubnikv, 2020-12-03:
> I still find it very weird that windows has absolutely no problem ejecting the device, but programs do.

Because I suppose a different API is used. Windows differentiates between unmount and eject, the API does one thing and the Explorer UI does another thing. This was clear when trying to detect plug / unplug events using various Win32 APIs: Unmounting using PrusaSlicer or Cura did not produce an Eject event to be captured back by PrusaSlicer while eject with Windows Explorer did produce such an event. That is about I know about the differences, it is not very well documented.

leptun, 2020-12-03:
PrusaSlicer seems to keep file handles open long after it wrote the file. They do close after a minute or so:
[image]

And whenever I eject the card from PrusaSlicer I get these files opened on the card by other processes:
[image]

So something is happening, but not successfully. Does this information help you in any way?

bubnikv, 2020-12-03:
> PrusaSlicer seems to keep file handles open long after it wrote the file.

I quickly skimmed over the code and it seems fine.

> They do close after a minute or so:

Without any user interaction with PrusaSlicer?
That is really weird. 

What tool are you using for checking the open file handles?

leptun, 2020-12-03:
It is possible that the tool that I'm using (Resource monitor from window, disk activity section) caches the handles for a while so that you can read them. That would be my guess to why they stay "open" for so long.

bubnikv, 2020-12-03:
You may try the diskmon from sysinternals
https://docs.microsoft.com/en-us/sysinternals/downloads/diskmon

closed 2023-07-07 by leptun