prusa3d/PrusaSlicer #4946: Feature Request - allow copy and paste between instances or programs

neophyl, 2020-10-23:
### Version
_Version of PrusaSlicer used goes here_
2.3.0 alpha 1 but applies to earlier ones

### Operating system type + version
Windows 10 but should apply to all OS

### 3D printer brand / version + firmware version (if known)
N/A

At the moment with the 2.3.0 you can run multiple instances of the slicer.  However you cant copy and paste objects between those instances.

Would be a nice feature to have.

This came about from a post on the Prusa forum asking if it was possible to copy an stl in one program of from the File explorer and use Paste to paste it to the bed after fast switching between programs.  Unfortunately not.  So I tried with 2 instances of slicer with the same result.  Would be nice to allow cross program pasting too.

itrmc, 2024-09-23:
Would love to see this to help with slicing workflow for multiple printers.

jimisola, 2024-10-15:
Such a basic feature for any application on any OS.

pravic, 2026-09-02:
> any issue that is not a reproducible bug, will be closed in GitHub issue tracker. 

I couldn't find what does the policy say about the _feature requests_. Because spamming with "+1" on every feature request every 3 weeks in order to prevent it from closing is kinda useless and only would annoy the maintainers.

Jan-Soustruznik, 2026-10-02:
Hello @pravic and everyone, I believe this requested feature is now available. PrusaSlicer now supports internal tabs (similar to a web browser), so it is no longer necessary to open multiple instances. You can simply open a new tab and load or create another project. It is also now possible to copy and paste models, along with their settings, directly between tabs.

closed 2026-10-02 by Jan-Soustruznik