prusa3d/PrusaSlicer #4947: Excessive moves during large Lithophane prints

Doprintityourself, 2020-10-23:
### Version
V2.2.0 and 2.3.0Alpha

### Operating system type + version
Win10 64 bit

### 3D printer brand / version + firmware version (if known)
13 printers of various brands

### Behavior
Excessive movement takes enormous ways? 
(Avoid Crossing Parameter does not help and with this file size and type newer finishes slicing!)

Open file (cannot upload because of size) 

and slice:
Blank all but Movements

_Is this a new feature request?_
Yes Partially would like to have Time and % for values for movement to have a better interpretation.
Same for a count of retracts.

[image]

Excessive Movement:
I dont know if this is a specific to Lithpophane or Size or shape,
but is there no other way or feature to optimize much better the excessive Movcements

#### Project File (.3MF) where problem occurs

Cannot upload the file it is compressed more than 40 MB of gcode
is there any other way to upload?

neophyl, 2020-10-23:
what is the size of the .3mf project file when exported from Prusa Slicer ? Instead of the gcode.  Project file is usually more usefull anyway and what gets asked for.

Doprintityourself, 2020-10-24:
Even if zipped 3mf (63MB)file has about 40MB's so no way to upload here.
If somebody will have a look,  I would really appreciate and transfer it to a reachable gdrive for example
Thanks

Doprintityourself, 2020-10-24:
so as it was easy to do I just uploaded the zip here:
https://drive.google.com/file/d/1eboHWTzytBChi0Cesoz9fpv3r8PKJmji/view?usp=sharing

bubnikv, 2020-12-07:
I think it is the gap fill.

closed 2026-09-29 by prusaslicer-bot[bot]