prusa3d/PrusaSlicer #4948: Concentric Fill Start Point - New Feature Request

iso250, 2020-10-23:
### Version

All

### Operating system type + version

All

### 3D printer brand / version + firmware version (if known)

All

### Behavior

New Feature Request

When you print an object with Concentric Fill the fill sometimes starts from the perimeter of the object, sometimes it starts from the middle.  It would be good if we could actively choose where it started from.

Why would this be a good feature?  Because when you print two identical objects at the same time using Concentric Fill one will start from the perimeter and the other will start from the middle, every time.  Even though the objects are printed using the same settings the sheen looks different on them, meaning they don't look uniform.  Often I need my prints to look exactly the same.

sg000, 2020-12-23:
I think this is very similar, if not identical to #5214 
As I pointed there:
Having a 3-way selection for (slicer optimized [today's default], inside-out, outside-in) for both Concentric and Archimedean infill options would resolve this quite easily for all of us.

sg000, 2020-12-23:
I will also note that somewhere between alpha 2 or 3 and rc2, the behavior has changed. It is always starting in the middle and working out, which IMOHO is not the best way. Its doesn't always eliminate the middle gap, AND it always creates the shortest paths first in a completely open area which are the hardest to keep adhered to the plate, rather than working off the perimeter which is typically more stable.

bubnikv, 2021-01-08:
Concentric infill will be printed outside-in in PrusaSlicer 2.3.0-rc3. This is consistent with the original Slic3r and it makes the most sense IMHO.

> I will also note that somewhere between alpha 2 or 3 and rc2, the behavior has changed.

That is just a bug. It should always have printed outside-in.

closed 2021-01-08 by bubnikv

iso250, 2021-01-08:
This is bad news. When it fills from the outside to the middle it often leaves big blobs and scars where it tries to squeeze in a line to fill the last gap. 

How about an option to fill in a particular direction for those of us that like the clean finish inside to outside provides?

(For a lot of my prints I’ll need to keep the beta as outside to inside is essentially broken, making the new release useless to me.)