prusa3d/PrusaSlicer #4949: SD Card Eject Button Not Present After Export

RedBullEnema, 2020-10-23:
### Version
2.3.0-alpha1 + win64

### Operating system type + version
Win10, Build 2004.

### 3D printer brand / version + firmware version (if known)
MK3S Clone / Marlin Bugfix-2.0.x

### Behavior
After using "Export to SD Card / Flash Drive", the eject button is not presented/does not appear. 

_Is this a new feature request?_
No, appears to be a bug.

kocikdav, 2020-10-25:
Hello, the eject button has been moved to the notification in form of clickable text. It should appear on the right bottom side of plater screen. Please try alpha2 as there were some bugs repaired.

fe60, 2020-10-28:
@RedBullEnema is aware of the link in the notification pop-up, but doesn't like it.

The option to take action from a notification is a nice addition. But it should not replace a (well known) button.
An eject button is easy to understand (similar to physical eject buttons with a similar icon) and the existing one was easy to find.

What is the idea behind the changed behaviour?
In which way does this link in the notification pop-up improve the usability compared to the button?

bubnikv, 2020-10-28:
One reason was that the eject was too close to the other buttons and one of
our slicer power users was consistently hitting that button by mistake.

RedBullEnema, 2020-10-28:
@fe60 
Hit the nail on the head, the eject button was big, easy, and everyone knows where it is. To be honest I didn't actually notice the text popup on the platter screen initially, completely missed it because notifications like that have never been used before (that Ive seen).   

@bubnikv 
I'm sorry, and I totally understand the value of power users, but that sounds like more of a user error than UI problem. This program is COVERED in tiny tick boxes and text fields (in fact, thats almost the entirety of it), and you're getting rid of one of the largest buttons in it? 
For the sake of argument, lets pretend that 25% of users are somehow accidentally hitting this big ole button.. a button that doesn't even appear until the slice is exported to the SD, at which point I would assume the majority of people intend to eject the SD, lets say 75%, so now your user group shrinks significantly. 

I really do love this program, its IMHO the best out there right now (and somehow free), but this just seems like the opposite of a logical reason to completely remove a large and important tool on the UI. Make the button bigger, reposition it to the other side of the "Export G-code" button, above or below, etc. Completely removing it and changing the UI simply makes _**no**_ sense to me. It just blows my mind as lacking solid logic for such a large UI change.   

Seriously though, I love you guys and appreciate your hard work, thank you for receptive to discussion.

fe60, 2020-10-28:
I see two things that are not perfect about the eject button:
1. The eject button is close to the export button ("harmful" operation right next to a "harmless" one).
2. The UI area on the right side of the large "Slice now" button (marked with the red box) changes its behaviour due to new buttons that appear after slicing or export to SD card has been done.
[image]

Instead of completely removing the button, it could be rearranged (as proposed by @RedBullEnema).
Here is a proposal from my side:
1. Add a slightly larger separator between export and eject.
2. Always show the "Export to SD" and the "Eject" button, but make them inactive, when they can't be used.
3. Add a preference (don't know where) to toggle the visibility of the SD card related buttons, for people not using SD cards at all.

[image]
Slice
[image]
Export to SD card
[image]
Eject SD card
[image]

derenma, 2020-10-29:
@fe60 I am seeing this issue again with alpha2. (It was noted in the changelogs that this was fixed, but alas, the issue returns.)

fe60, 2020-10-30:
@derenma I don't konw if it was broken in alpha1, but in alpha2 the newly introduced notification appears in Simple mode and the "Eject drive" link does what it is supposed to to. The point of this issue is the replacement/removal of the eject button.

[image]

foreachthing, 2020-10-30:
> The point of this issue is the replacement/removal of the eject button.

I really liked the eject button. I miss it... :-(

fe60, 2020-10-30:
Just realized, that after an export to SD card the menu item and the short cut for ejecting a SD card are disabled. I guess this are side effects of the removal of the eject button.

[image]

fe60, 2020-11-15:
The export notification got a kind of an eject button in 2.3.0-alpha3.
[image]
I still don't like it, but I started using short cuts in Prusa Slicer and I don't care anymore about the design of the eject button.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha3
_Reworked Export finished notification. It now contains path to G-code, "Open folder" hyperlink and Eject removable media button._

fe60, 2020-11-22:
The eject button in the notification became much bigger in 2.3.0-alpha4
[image]

@RedBullEnema What do you think about the new design?

bubnikv, 2020-12-07:
This is killing me.
I have asked @mikolaszuza to open voting on twitter.
Or do you guys want to make it configurable?
Personally I don't care, both work for me.

fe60, 2020-12-07:
Since the button in the notification has been made larger in alpha4 it is fine for me.
But as mentioned on [November 15h](https://github.com/prusa3d/PrusaSlicer/issues/4949#issuecomment-727554482), I switched to shortcuts.

@bubnikv The shortcut is not shown in the tool tip of the new eject button. Showing it there (as it was the case for the "old" button) will make it easier for users, to become aware of the shortcut.

fe60, 2021-11-09:
I got used to the Eject button placed in the notification. Just the tool tip should be shown earlier (#7278).

@RedBullEnema and @foreachthing : What about you? Can this ticket be closed?

lukasmatena, 2021-11-09:
The change has been in there for more than a year now, there were no other complains as far as I remember. I believe this can indeed be closed.

closed 2021-11-09 by lukasmatena