prusa3d/PrusaSlicer #4950: All Top Surfaces Ironing Issue

FrenzyIncarnate, 2020-10-23:
### Version
2.3.0 Alpha-1

### Operating system type + version
Windows10 64bit
### 3D printer brand / version + firmware version (if known)
Prusa Mini Fw 4.2.1
### Behavior
As can be seen in the pictures after ironing surfaces in the mid level of the model there is a line at the ironing level like printer overextruded or skipped a step, creating bad perimeter.
* _Steps needed to reproduce the problem_
All top surfaces option needs to be selected.
* _Expected Results_
There shouldnt be any visual deformation at the surface
* _Actual Results_
There are lines at the mid ironing layers where the shape continues .
 
3mf file link (wetransfer) https://we.tl/t-nXFqBeCAc5
[image]
[image]

bubnikv, 2020-12-07:
Would you please zip the 3md and attach it here? Thanks.

michael-kerbel, 2020-12-13:
this might be connected to #5113

FrenzyIncarnate, 2020-12-13:
> Would you please zip the 3md and attach it here? Thanks.

I really am sorry but I have lost all the files since I've opened this 1.5 months ago and I accept that wetransfer wasnt the best location to upload the files.

bubnikv, 2020-12-16:
OpenSCAD code to produce model similar to yours
```
difference() {
    cube([60, 60, 20]);
    translate([20, 0, -0.5]) cube([40, 40, 21]);
};
translate([0, 20, 0]) cube([20, 40, 40]);
translate([0, 40, 0]) cube([20, 20, 60]);
```

bubnikv, 2020-12-16:
[stairs.zip](https://github.com/prusa3d/PrusaSlicer/files/5701426/stairs.zip)

bubnikv, 2020-12-16:
Ironing is not being done inside the object.

[image]
[image]

michael-kerbel, 2020-12-16:
The reporter never claimed, that ironing was being done inside. The issue is, and I can confirm this, that after doing lots of ironing on a layer, the perimeters on the next layer are „fat“. I guess, it happens due to the very low flow of filament during the ironing.

michael-kerbel, 2020-12-17:
I found a manual workaround for the issue of fat layers: https://github.com/prusa3d/PrusaSlicer/issues/5113#issuecomment-747602704

closed 2026-09-29 by prusaslicer-bot[bot]