prusa3d/PrusaSlicer #4953: [feature request]

userosos, 2020-10-24:
Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 16,903MB

https://imgur.com/a/sO7IGK4

I want there to be a Z = 0.1 mm gap between the support interface layer and the next model layer, with no bridge infill between the support interface and the model.
[Test supports.zip](https://github.com/prusa3d/PrusaSlicer/files/5432807/Test.supports.zip)

userosos, 2020-10-24:
Now i can see that if i set Contact Z distance to 0.1 mm. In g-code i can see Contact Z distance = ~0.3 mm? It is OK? I can't set Contact Z distance to 0.1 mm in gcode. It is bug?

userosos, 2020-10-24:
I set Contact Z distance to -0.1 mm it work for 0.2 layer. If i set layer to 0.3 for example i set to 0.1 mm. If i use addaptiv lauers it not work. It is bug?

rtyr, 2020-10-24:
Expected behavior, this is how PrusaSlicer (and upstream Slic3r) works. The bottom most surface of the overhang is printed with a bridge flow. Bridge is round extrusion with diameter equal to the nozzle diameter (0.4mm in this case).

You can find more information on how PrusaSlicer works with the supports here #102 (there are also some duplicate issues linked to 102). It is possible to use negative values for z-offset if you need smaller distance. Closing.

[image]

closed 2020-10-24 by rtyr