prusa3d/PrusaSlicer #4955: PrusaSlicer 2.3.0 alpha1 will not load on Macbook Pro

oldgit9, 2020-10-24:
### Version
2.3.0 alpha1

### Operating system type + version
OSX 10.15.7 

### Behavior
* _Nothing happens when i click on the PrusaSlicer icon.
* _Move/copy the application from the .dmg folder to the Applications folder.
* _Expect PrusaSlicer to open, when i double click on the icon.
* _Nothing happens, program does not open.

neophyl, 2020-10-24:
Have you read https://github.com/prusa3d/PrusaSlicer/issues/4889 ?

oldgit9, 2020-10-24:
> Have you read #4889 ?

thanks for the link. yes i have and unfortunately for me, it didnt work.

rtyr, 2020-10-25:
Can you please try it with 2.3.0-alpha2?

oldgit9, 2020-10-25:
> Could you please try it with 2.3.0-alpha2?

thanks for your reply. i can confirm, alpha2 does work on my macbook. time to experiment with the new features.

rtyr, 2020-10-25:
Thanks. Closing.

closed 2020-10-25 by rtyr