prusa3d/PrusaSlicer #4956: Continue infill when geometry changes

fcore117, 2020-10-24:
2.3.0 a1
Win 10 x64

Continue infill, i am new to 3D printing so images might explain. One is fom PrusaSlicer and other is from Cura and would like functioning like in Cura. Close it if there is similar isue or there is already option for this and i missed that.

[image]
[image]

_Is this a new feature request?_
Actually dont know yet.

fcore117, 2020-10-26:
I hope final 2.3.0 will get improved in this area if it really is not my own mistake. Please slice it with PS and Cura.
[ww_warning_Triangle_kbob1.zip](https://github.com/prusa3d/PrusaSlicer/files/5436163/ww_warning_Triangle_kbob1.zip)

fcore117, 2020-11-05:
Large things can continue infill but those edges where it continues seems to be have much less stability.

fcore117, 2020-11-05:
Large objects infill that is created seems to have less stability if i look sliced comparision, Cura seems to build some sort of neck when continuing building after geometry change. 1 Cura(yellow) 2 PS (red)
[image]
[image]

[geometry_sample_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/5493995/geometry_sample_v2.zip)

bubnikv, 2020-12-07:
Isn't it a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/89 ?

closed 2026-09-02 by fcore117