prusa3d/PrusaSlicer #4957: Monotonic fill defaulting to rectilinear suddenly

Al3Dx, 2020-10-24:
### Version
2.3.0 alpha

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
TT Sapphire Pro/Marlin 2.0

### Behavior
Monotonic infill worked fine and produced great results, since yesterday it started printing rectilinear instead and I can't get it working again. Also tried an older snapshot. Concentric and all still works.

Great work on the new version, loving it!

rtyr, 2020-10-24:
Please provide your gcode and project file (3MF).

Al3Dx, 2020-10-24:
[3DDC_Testcube_V2.zip](https://github.com/prusa3d/PrusaSlicer/files/5433091/3DDC_Testcube_V2.zip)

Al3Dx, 2020-10-24:
This is what my preview shows with monotonic selected

[image]

rtyr, 2020-10-24:
Your gcode is OK, it is printed with monotonic infill.

I don't see anything wrong on your screenshot.

rtyr, 2020-10-24:
How monotonic infill works https://github.com/prusa3d/PrusaSlicer/releases. You can compare the preview with rectilinear/monotonic to see the difference between classic rectilinear and monotonic.

rtyr, 2020-10-24:
Working as expected. Closing.

closed 2020-10-24 by rtyr

Al3Dx, 2020-10-24:
I misunderstood the functionality, truly sorry for that.

rtyr, 2020-10-24:
No problem :-).