prusa3d/PrusaSlicer #4959: Cannot build "cmake -DSLIC3R_WX_STABLE=1" (Could NOT find wxWidgets)

jarl-dk, 2020-10-24:
### Version
git version b8e77e4d3

### Operating system type + version
Ubuntu 20.04

### Behavior
`cmake -DSLIC3R_WX_STABLE=1 ..` fails with

* _Describe the problem_
```
git checkout b8e77e4d3
mkdir build
cd build
cmake -DSLIC3R_WX_STABLE=1 ..
```

* _Expected Results_
cmake will finish without errors

* _Actual Results_
The output is
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: 1
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found DBUS: /usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time 
-- Boost::boost exists
-- Found TBB: /usr/include (found version "2020.1")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /usr/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /usr/lib/x86_64-linux-gnu/libtbb.so
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /usr/lib/x86_64-linux-gnu/libtbb.so
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.68.0") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3") 
-- Found EXPAT: /usr/lib/x86_64-linux-gnu/libexpat.so (found version "2.2.9") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.37") 
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.1.0") 
-- Could NOT find cereal (missing: cereal_DIR)
-- Looking for C++ include cereal/cereal.hpp
-- Looking for C++ include cereal/cereal.hpp - found
-- Found NLopt in '/usr/lib/x86_64-linux-gnu'.
-- Using NLopt include directory '/usr/include'.
-- Using NLopt library '/usr/lib/x86_64-linux-gnu/libnlopt.so'.
-- Found OpenVDB: /usr/include (found suitable version "6.2.1", minimum required is "5.0") found components: openvdb 
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:296 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:449 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:303 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "6" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:449 (find_package)

-- Falling back to IlmBase found by pkg-config...
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: iostreams system 
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
-- OpenVDB libraries: /usr/lib/x86_64-linux-gnu/libopenvdb.so
-- Miniz NOT found in system, using bundled version...
-- qhull Version: 7.2.0 (static linking)
-- IGL NOT found, using bundled version...
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmp.so  
-- Found MPFR: /usr/lib/x86_64-linux-gnu/libmpfr.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.48")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Using gcc version 4 or later. Adding -frounding-math
-- Adding precompiled header pchheader.hpp to target libslic3r.
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS) (Required is at least version "3.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindwxWidgets.cmake:1008 (find_package_handle_standard_args)
  src/CMakeLists.txt:43 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jarl/projects/PrusaSlicer/build/CMakeFiles/CMakeOutput.log".
See also "/home/jarl/projects/PrusaSlicer/build/CMakeFiles/CMakeError.log".

```
`CMakeFiles/CMakeError.log` contains:
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/jarl/projects/PrusaSlicer/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_3fbf3/fast && /usr/bin/make -f CMakeFiles/cmTC_3fbf3.dir/build.make CMakeFiles/cmTC_3fbf3.dir/build
make[1]: Går til katalog "/home/jarl/projects/PrusaSlicer/build/CMakeFiles/CMakeTmp"
Building C object CMakeFiles/cmTC_3fbf3.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD -fPIE   -o CMakeFiles/cmTC_3fbf3.dir/src.c.o   -c /home/jarl/projects/PrusaSlicer/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_3fbf3
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_3fbf3.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    CMakeFiles/cmTC_3fbf3.dir/src.c.o  -o cmTC_3fbf3 
/usr/bin/ld: CMakeFiles/cmTC_3fbf3.dir/src.c.o: in function `main':
src.c:(.text+0x46): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_3fbf3.dir/build.make:87: cmTC_3fbf3] Fejl 1
make[1]: Forlader katalog "/home/jarl/projects/PrusaSlicer/build/CMakeFiles/CMakeTmp"
make: *** [Makefile:121: cmTC_3fbf3/fast] Fejl 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

```

_Is this a new feature request?_
No

@AJRepo has opened a similar issue at https://github.com/prusa3d/PrusaSlicer/issues/4782 but has closed it with no good reason, therefor I open a new issue.

AJRepo, 2020-10-24:
I closed it because I could get it to compile with statically linked wxwidgets. See comment:

> I can get cmake to run and use make to compile if I use a statically linked wxwidgets-3.1 (e.g. cmake -DOpenGL_GL_PREFERENCE=GLVND -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH=/path/to/destdir/usr/local -DSLIC3R_STATIC=1 .. )

jarl-dk, 2020-10-25:
@AJRepo : OK, thanks for theinformation, it was not clear to me from the last commentsin that issue. Anyway that is a workaround, right?

This bug report is about cmake fails with default (dynamcally linked) wxwidgets.

lukasmatena, 2020-10-25:
> This bug report is about cmake fails with default (dynamcally linked) wxwidgets.

Actually, linking wxWidgets statically is the recommended way, we in fact use a slightly modified version of wxWidgets 3.1 with certain backported features from current master. We have even made nice set of cmake scripts to automate building the dependencies. That is how our builds are produced and how it is more or less guaranteed to work.

Linking dynamically is also possible, but the responsibility about whether cmake finds all required libraries and their compatible versions is on anyone trying to do that. I can only advise to turn off precompiled headers, which do not bring much benefit on Linux, except from breaking the build quite often in unpredictable way (I think the pthread problem could be caused by them). Just add `DSLIC3R_PCH=0` to cmake arguments.

jarl-dk, 2020-11-14:
@lukasmatena : Thank you very much. Maybe the the How to biuld file should be updated with the information about the recommendation about static.
Running `cmake .. -DSLIC3R_STATIC=1` gives another problem reporter earlier on https://github.com/prusa3d/PrusaSlicer/issues/4184 which is cloesd without a description of the resolution.
```shell-script
$ cmake .. -DSLIC3R_STATIC=1
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: 1
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: 1
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found DBUS: /usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time 
-- Boost::boost exists
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version "2020.1")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake:286 (find_package_handle_standard_args)
  CMakeLists.txt:328 (find_package)

-- Configuring incomplete, errors occurred!

```

I do have tbb development package installed:
```shell
$ LANG=C aptitude search tbb
p   libbtbb-dev                                 - Bluetooth baseband decoding library - development fil
p   libbtbb1                                    - Bluetooth baseband decoding library                  
p   libmkl-tbb-thread                           - Intel? MKL: Intel TBB threading library for the Intel
i   libtbb-dev                                  - parallelism library for C++ - development files      
p   libtbb-dev:i386                             - parallelism library for C++ - development files      
p   libtbb-doc                                  - parallelism library for C++ - documentation          
v   libtbb-doc:i386                             -                                                      
i A libtbb2                                     - parallelism library for C++ - runtime files          
p   libtbb2:i386                                - parallelism library for C++ - runtime files          
```

Any help is appreciated.

bver, 2020-11-21:
Observing the exactly same behavior as @jarl-dk does.
Could you please address this issue? Thanks.

bver, 2020-11-22:
After some struggle I've managed to compile the Slicer with wxWidgets 3.1 instead. 
The key point was to realize that there is a `v3.1.1-patched` branch:
https://github.com/prusa3d/wxWidgets
available which is not considered as a dependency by default. Therefore this is important to clone or link this repo to `deps/` dir before the _Dependency resolution_ step is performed (not mentioned in doc though). Also, cmake caches your options so frequent cleaning `build/` dir helps a lot. 

The main cmake command which eventually worked for me:
`cmake .. -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=/home/pavel/work/3d/PrusaSlicer/deps/build/destdir/usr/local -DCMAKE_CONFIGURATION_TYPES=Release -DSLIC3R_PCH=0`

bubnikv, 2020-12-07:
Is it resolved? Would you please close then?
Contributions to the https://github.com/prusa3d/PrusaSlicer/wiki
are welcome.

jarl-dk, 2020-12-07:
What do you mean it is resolved? I just pulled the latest master and the source code cannot build according to instructions at https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

smirgol, 2021-01-16:
So, I've messed with getting this to compile for hours and finally got it working.
Start with a fresh clone of the repo, change directory to that clone.

Build dependencies:
`cd deps && mkdir build && cd build`
`cmake ..`
`make -j4` 

If that works, go back to the root folder and create another build dir:
`cd .. && mkdir build && cd build`

Then do the cmake stuff:
``cmake .. -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=<absolute(!)_path_to_deps>/build/destdir/usr/local/ -DCMAKE_BUILD_TYPE=Release -DSLIC3R_GTK=2  -DSLIC3R_WX_STABLE=0``

If that works, compile:
`make -j4`

If it complains about other missing libraries, install them.

I could not get it to work with my system installation of wxWidgets at all. It would also not work with the deps version of wxWidgets and the culprit turned out to be the GTK version. It defaults to version 3 on my system and I explicitly need it to tell to use version 2. Then it will build with wxWidgets 3.0 and also 3.1. The version supplied by the deps seems not to support gtk3.
`

nordurljosahvida, 2021-08-24:
@bver could you described how you statically linked wxwidgets 3.1? In my deps dir: `cat wxWidgets/wxWidgets.cmake` I see in the first line: `set(_wx_git_tag v3.1.4-patched)`. Does this mean I have them already linked?

nordurljosahvida, 2021-08-24:
@bubnikv I've made a "building on ubuntu 20" guide with all the steps mentioned here: https://github.com/prusa3d/PrusaSlicer/issues/3099. Do note that vojtechkral didn't necessarily want ready-to-run instructions so the debate is open on #3099. I do have to fix the GTK3 issue and then the guide is ready.

lukasmatena, 2021-09-02:
@jarl-dk I have updated the build instructions to include complete set of commands that are needed to build it. I have tested them and they work. Can we close the issue now?

nordurljosahvida, 2021-09-02:
@lukasmatena thanks for updating, I still really think it would be great to have a step-by-step guide as well beyond the documentation. I've updated my guide here: https://github.com/prusa3d/PrusaSlicer/pull/6831, if we could please have a reference, this is the document I refer to every time I build. Also, I'm using ninja as per https://github.com/prusa3d/PrusaSlicer/issues/3099#issuecomment-548268917, not sure if it makes any difference, but it works.

lukasmatena, 2021-09-02:
@nordurljosahvida 
> I still really think it would be great to have a step-by-step guide 

Actually, I thought I was writing exactly that, I even named the chapter 'Step by step guide' and verified that running the commands one by one without thinking built PrusaSlicer. Its contents is in fact the same as your document, with several descriptive sentences in between. Frankly, I don't understand what else might be needed, given that the whole build really consists of two cmake invocations...

nordurljosahvida, 2021-09-02:
@lukasmatena I must have been reading a cached version for some reason, I hadn't seen all the latest additions and improvements! Thanks for implementing those and sorry for the noise.

lukasmatena, 2021-09-02:
Ok, i understand :-) If you have any more suggestions what should be mentioned to make it as clear as possible, let me know.

nordurljosahvida, 2021-09-02:
I think it's crystal clear.

A few extra questions if / when you have a minute:

- any benefit to building with ninja?
- `libgtk2.0-dev libudev-dev ninja-build` are needed and if so installed when building deps?
- don't remember if anyone had written a reply on how to build appimage?

lukasmatena, 2021-09-02:
> any benefit to building with ninja?

I don't know. I use it myself but when I tested whether it is really faster than make, the difference was negligible, even with incremental builds. I chose make for the guide simply not to force people install another package.

> libgtk2.0-dev libudev-dev ninja-build are needed and if so installed when building deps?

This is something we plan to add to the guide. Several packages are needed and NOT installed by the deps script (on Debian for a GTK3 build, those would e.g. be `libgtk-3-dev libglu1-mesa-dev` and maybe others). It is quite difficult to include this in the simple guide, because the package names are different under different distros and each distro comes with different set of packages preinstalled. The Linux world is nice, but it is also quite a mess, due to the freedom that everyone has. Right now it is just casually mentioned in the 'Troubleshooting' paragraph.

> don't remember if anyone had written a reply on how to build appimage?

Definitely nobody from the development team, and I don't think we would spend time on it. [AppImageKit](https://github.com/AppImage/AppImageKit) is the tool used for that, but we cannot provide any assistance.

Closing the issue.

closed 2021-09-02 by lukasmatena

bubnikv, 2021-09-03:
- any benefit to building with ninja

It is supposed to be faster and it runs a parallel compilation by default
(in contrast to make).

jarl-dk, 2021-11-07:
Thanks @lukasmatena. Sorry for not responding to this earlier.
The build documentation is much improved. However the build steps fails (only tried on Ubuntu 21.10).
Even though I have `libgtk-3-dev` and `libcairo2-dev` installed it fails during building of dependancies finishing with

```shell
/home/jarl/projects/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/cairo.cpp:24:10: fatal error: cairo.h: No such file or directory
   24 | #include <cairo.h>
      |          ^~~~~~~~~
compilation terminated.
gmake[5]: *** [libs/core/CMakeFiles/wxcore.dir/build.make:287: libs/core/CMakeFiles/wxcore.dir/__/__/__/__/src/common/cairo.cpp.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
/home/jarl/projects/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/generic/graphicc.cpp:29:10: fatal error: cairo.h: No such file or directory
   29 | #include <cairo.h>
      |          ^~~~~~~~~
compilation terminated.
gmake[5]: *** [libs/core/CMakeFiles/wxcore.dir/build.make:791: libs/core/CMakeFiles/wxcore.dir/__/__/__/__/src/generic/graphicc.cpp.o] Error 1
In file included from /home/jarl/projects/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/combocmn.cpp:162:
/home/jarl/projects/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/gtk1/private.h:14:10: fatal error: gdk/gdk.h: No such file or directory
   14 | #include <gdk/gdk.h>
      |          ^~~~~~~~~~~
compilation terminated.
gmake[5]: *** [libs/core/CMakeFiles/wxcore.dir/build.make:399: libs/core/CMakeFiles/wxcore.dir/__/__/__/__/src/common/combocmn.cpp.o] Error 1
gmake[4]: *** [CMakeFiles/Makefile2:704: libs/core/CMakeFiles/wxcore.dir/all] Error 2
gmake[3]: *** [Makefile:149: all] Error 2
make[2]: *** [CMakeFiles/dep_wxWidgets.dir/build.make:137: dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:122: CMakeFiles/dep_wxWidgets.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

```

I have made a minimalistic vagrant box for verifying the build steps on https://github.com/jarl-dk/PrusaSlicerBuildBox

lukasmatena, 2021-11-08:
@jarl-dk I did not run your build, but if you want to build against GTK3, you should set `-DDEP_WX_GTK3=ON` when building dependencies and `-DSLIC3R_GTK=3` when building PrusaSlicer. Whether it will be enough to solve the problem, I don't know.

jarl-dk, 2021-11-08:
Hi @lukasmatena 
Thanks for getting back

I don't have any particular wishes to build against anything in particular. I am just trying to follow your build steps: 
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

What should I do if I _do not_ have any particular wishes to build against GTK3 (or any other specific library) and just want to follow the default build steps?

If there is any information missing in the Linux build document, please update it. If there are any particular packages that is needed prior to building, please add that to the documentation as well.

lukasmatena, 2021-11-08:
@jarl-dk 
>  If there are any particular packages that is needed prior to building, please add that to the documentation as well.

I suppose I may add packages that are possibly missing for GTK3 builds on Debian/Ubuntu, given that it is very widely used distro. However, I do not plan to install all imaginable Linux distributions, search for packages and list the names needed for both GTK2 and GTK3 variants.

Can you confirm that when you do what I said and install `libgtk-3-dev` and `libglu1-mesa-dev`, the build is fine? Thanks.

jarl-dk, 2021-11-13:
I have now successfully built it from the documentation.
It was not at all obvious to figure out which dependancies was needed for the buld.
I have successfully identified the minimal buld requirements (for Ubuntu), you can see them here:
https://github.com/jarl-dk/PrusaSlicerBuildBox/blob/cd8ee546c5addbb4d5c43496ea1693db79c6caca/Vagrantfile#L77-L90

I would suggest to list the required libraries to build prusa slicer. You have only mentioned some of them. Even worse is that the documents says that `libgtk-3-dev` is needed , but the later provided build command requires `libgtk2.0-dev` (not libgtk 3).

jarl-dk, 2021-11-13:
I have created a PR to make the build-requirements more complete (for Ubuntu as an example): https://github.com/prusa3d/PrusaSlicer/pull/7302