prusa3d/PrusaSlicer #4961: 2.3.0-alpha2: Uncaught exception: Calling AppConfig::save() from a worker thread!

probonopd, 2020-10-24:
`PrusaSlicer-2.3.0-alpha2+linux-x64-202010241644.AppImage` on `ubuntu-18.04-desktop-amd64.iso`

[image]

```
[2020-10-24 20:53:29.371931] [0x000000080211c2c0] [error]   Uncaught exception: Calling AppConfig::save() from a worker thread!
Unhandled unknown exception; terminating the application.
```

bubnikv, 2020-10-25:
When did it happen?

I did two changes in PrusaSlicer 2.3.0-alpha2:
1) Some important threads are given names. The main thread is slic3r_main
2) Only the slic3r_main thread is allowed to save the PrusaSlicer.ini

The test 2) is there to confirm a hypothesis of why very rarely PrusaSlicer.ini gets corrupted on Windows. I wonder what went wrong on your system, when and why. Is the function really called by a worker thread? Or was the main thread not given its name, or did reading out the name of the current thread fail?

probonopd, 2020-10-25:
To be honest, I have no clue. I just tried to run the AppImage.

bubnikv, 2020-10-26:
> To be honest, I have no clue. I just tried to run the AppImage.

But when did it happen? Just at the start? Or you don't remember anymore?

probonopd, 2020-10-27:
I got this at application start.

bubnikv, 2020-10-27:
Is it reproducible?

probonopd, 2020-10-28:
Yes, when I had tried 2 days ago I had been unable to start the application, also when trying multiple times. (I have no access to that machine right now.)

bubnikv, 2020-12-07:
Do you still have it with beta2? I find it weird that nobody else reported this issue.

probonopd, 2020-12-07:
Looks like I no longer get this error. Thank you very much @bubnikv

closed 2020-12-07 by probonopd

christina002, 2021-04-14:
I just ran into this exception with PrusaSlicer 2.3.0 (release). I'm running the Linux binary on FreeBSD's Linux compatibility layer so there might be differences in thread preemption; the Linux shared libraries are from CentOS 7. It works fine in general but runs into this exception as soon as I do anything (e.g. click on the "+" toolbar button to add an STL object, then Cancel), then try to close the program.

bubnikv, 2021-04-14:
On FreeBSD, the issue may be the main thread identification. I cannot know
without actually debugging it.
We may disable the check on FreeBSD, but I suppose you are running a Linux
binary compiled by us?